MichaluxPL/Sofar_LSW3 #3: zmienna totalpower i mqtt

pablolite, 2021-09-30:
Zauważyłem dwa małe błędy.

1. zmienna "totalpower" (zapewne też totaltime) po wyjściu z pętli mają wartość = 0
To chyba problem z zasięgiem czy wcięciem. Nie znam się na programowaniu,
ale jak zadeklarowałem ja na początku kodu (bez wcięcia) to działa prawidłowo.
2. Broker MQTT nie dostawał prawidłowych danych. Tzn. otrzymywał tylko komunikat o "totalpower".
Jak zmieniłem kolejność publikacji to teraz trzyma dane o atrybutach i totalpower.
 client.publish(mqtt_topic+"/attributes",output)
 client.publish(mqtt_topic,totalpower)

MichaluxPL, 2021-09-30:
Ad1. Rzeczywiście był błąd, który generalnie wpływał tylko na wartość przekazywaną do MQTT. W najbliższej wersji będzie poprawione.
Ad2. MQTT to fragment oryginalnego kody jlopez77. Nie testowałem tego.

pablolite, 2021-09-30:
Ad2.
Doczytałem w dokumentacji paho.mqtt, że w ten sposób nie można puścić wielu publish do brokera.

Zmieniłem u siebie fragment kodu na:
client.connect(mqtt_server, mqtt_port)
client.loop_start()
client.publish(mqtt_topic+"/attributes",output)
client.publish(mqtt_topic,totalpower)

Teraz kolejność i ilość publish nie ma znaczenia, można ich wykonać kilkanaście.

Przez kilka dni spróbuje opracować publikowanie danych do brokera w taki sposób, że "home assistant"
wykryje go automatycznie przez integracje MQTT. Jak się uda to podzielę się zmianami w kodzie, może wykorzystasz :)

MichaluxPL, 2021-09-30:
OK, jak przetestujesz, to podeślij kod, to wrzucę :)

pablolite, 2021-10-03:
Cześć,

Nie wiem jak wrzucić kod, żebyś zobaczył zmiany.
Nie korzystam z tego typu narzędzi i nie programuje :)

Efekt działania jest taki, że Home Assistant sam "zauważa" nowy logger i
zbiera jego odczyty z brokera MQTT.

[image: image.png]
[image: image.png]
[image: image.png]

Całkowita produkcja i czas pracy występują dwa razy, bo całość odczytuje z
pętli i chyba dwa razy to musi gdzieś występować w odczytach.
Nie obsłużyłem jeszcze kodu kraju, bo nie bardzo wiem jak się do tego
dobrać.

Dodałem tutaj do załącznika.
Generalnie zrobiłem 3 zmiany jak poniżej:

# END CONFIG
totalpower=0
totaltime=0
*odczyty = []*
(...)
     for register in item["registers"]:
      if register==hexpos and chunks!=-1:
       response=round(response*ratio,2)

*#wyniki do tablicy w celu integracji MQTT z HomeAssistant
 odczyty.append([title, ratio, unit, metric_type, metric_name, label_name,
label_value, response,register])*
       for option in item["optionRanges"]:
        if option["key"] == response:
            if lang == "PL":
(...)
  client.username_pw_set(username=mqtt_username, password=mqtt_passwd)

*# wysylanie danych do MQTT z automatycznym wykrywaniem sensorow w
HomeAssistant licznik=0 client.connect(mqtt_server,
mqtt_port) client.loop_start()
client.publish("sofar/sofar_logger/"+str(inverter_sn)+"/enabled","true")
client.publish("sofar/sofar_logger/"+str(inverter_sn)+"/state/connected","true")
for
odczyt in odczyty:# sensor "energy" dla jednostek produkcji pradu   if
odczyt[2]=="kWh" or odczyt[2]=="Wh" or odczyt[2]=="W":
client.publish("homeassistant/sensor/sofar_logger/"+str(inverter_sn)+"_"+str(licznik)+"/config","{\"avty\":{\"topic\":\"sofar/sofar_logger/"+str(inverter_sn)+"/state/connected\",\"payload_available\":\"true\",\"payload_not_available\":\"false\"},\"~\":\"sofar/sofar_logger/"+str(inverter_sn)+"/\",\"device\":{\"ids\":\""+str(inverter_sn)+"\",\"mf\":\"Sofar\",\"name\":\"WLS-3\",\"sw\":\"x.x.x\"},\"name\":\""+(odczyt[0])+"
["+(odczyt[2])+"]\",\"uniq_id\":\""+str(inverter_sn)+"_"+str(licznik)+"\",\"qos\":0,\"unit_of_meas\":\""+(odczyt[2])+"\",\"stat_t\":\"~state/"+(odczyt[4])+(odczyt[6])+"\",\"val_tpl\":\"{{
value | round(5)
}}\",\"dev_cla\":\"energy\",\"state_class\":\"total_increasing\"}")
client.publish("sofar/sofar_logger/"+str(inverter_sn)+"/state/"+(odczyt[4])+(odczyt[6]),(odczyt[7]))
  print ("["+str(licznik)+"]"+odczyt[0])    licznik=licznik+1# reszta
sensorow - mozna jeszcze jakies oddzielic   else:
client.publish("homeassistant/sensor/sofar_logger/"+str(inverter_sn)+"_"+str(licznik)+"/config","{\"avty\":{\"topic\":\"sofar/sofar_logger/"+str(inverter_sn)+"/state/connected\",\"payload_available\":\"true\",\"payload_not_available\":\"false\"},\"~\":\"sofar/sofar_logger/"+str(inverter_sn)+"/\",\"device\":{\"ids\":\""+str(inverter_sn)+"\",\"mf\":\"Sofar\",\"name\":\"WLS-3\",\"sw\":\"x.x.x\"},\"name\":\""+(odczyt[0])+"
["+(odczyt[2])+"]\",\"uniq_id\":\""+str(inverter_sn)+"_"+str(licznik)+"\",\"qos\":0,\"unit_of_meas\":\""+(odczyt[2])+"\",\"stat_t\":\"~state/"+(odczyt[4])+(odczyt[6])+"\",\"val_tpl\":\"{{
value | round(5)
}}\",\"dev_cla\":\"current\",\"state_class\":\"measurement\"}")
client.publish("sofar/sofar_logger/"+str(inverter_sn)+"/state/"+(odczyt[4])+(odczyt[6]),(odczyt[7]))
  print ("["+str(licznik)+"]"+odczyt[0])    licznik=licznik+1*
 print("Data has been sent to MQTT")
(...)

Pozdrawiam,
Paweł

MichaluxPL, 2021-10-03:
Trochę to mało przejrzyste - najlepiej zrób po prostu forka mojego projektu i nanieś swoje zmiany :)
Generalnie nie używam Home Assistanta, więc i tak tego nie przetestuję.

pablolite, 2021-10-04:
Ok wezmę to pod uwagę. Postaram się jakoś ogarnąć tego GIT'a i dodać zmiany.

Brakuje np. numeru wersji softu loggera i inwentera - da się to dodać?
W jaki sposób sprawdzasz adresy do konkretnych danych z loggera?

MichaluxPL, 2021-10-04:
Adresy są dostępne w dokumentacji (możesz takową pobrać z tego wątku na Elektrodzie: https://www.elektroda.pl/rtvforum/topic3698233.html -> plik załączony przez usera starob !)
Co do pobierania wersji softu itp, to jak najbardziej da się to pobrać, przy czym to wymaga wysłania komendy "4" zamiast "3" w nagłówku zapytania oraz dodania rozpoznawania kolejnych rejestrów. Ja już tego nie będę dodawał na razie, bo mi to nie potrzebne do monitorowania :)

pablolite, 2021-10-05:
Dzięki za podpowiedzi.
Czyli zmieniam:
businessfield= binascii.unhexlify('**0103**' + pos_ini + pos_fin) # sin CRC16MODBUS
na 
businessfield= binascii.unhexlify('**0104**' + pos_ini + pos_fin) # sin CRC16MODBUS

i czytam rejestry zgodnie z dokumentacją dla 0x04 ?

pablolite, 2021-10-05:
Jeżeli chodzi o obsługę MQTT Discovery w Home Assistant to zrobiłem poprawki, które działają.
https://github.com/pablolite/Sofar_LSW3/tree/HomeAssistantMQTT

Możesz zobaczyć co zmieniłem, pewnie nie jest to do końca ze sztuką programowania, 
bo robiłem to trochę metodą prób i błędów :)

pablolite, 2021-10-05:
Jak szukać właściwych wartości dla zmiennych pini, pfin?

MichaluxPL, 2021-10-05:
Prawdopodobnie zgodnie z adresacją rejestrów. Podstawowe rejestry zaczynają się od 0x0000 - więc pini ustawiłem na 0. Rozszerzone (manufacturer message) zaczynają się od 0x2000, więc obstawiam, że pini=8192. pfin to pewnie 8207 lub coś koło tego, sądząc z faktu, iż adres ostatniego rejestru to 0x2007. Można w sumie zczytać tylko do 0x200B (czyli pfin=8203), bo dalsze jak widać w dokumentacji są zarezerwowane, więc pewnie nic ciekawego nie przechowują.

closed 2021-10-21 by MichaluxPL