MichaluxPL/Sofar_LSW3 #10: MQTT

rysiulg, 2021-11-02:
Please add another config option:
"mqtt_ssl": "bool"
and in InverterData.py add mqtt_ssl config reader and in if to mqtt ssl (it's broken when ssl isn't used:
if mqtt_ssl==1:
 client.tls_set()  # <--- even without arguments

after that mqtt payload is looks like:

inverter/attributes
{"Inverter status":"Normal","Fault 1":"No error","Fault 2":"No error","Fault 3":"No error","Fault 4":"No error","Fault 5":"No error","PV1 Voltage (V)":301.0,"PV1 Current (A)":0.93,"PV2 Voltage (V)":302.7,"PV2 Current (A)":0.88,"PV1 Power (W)":280,"PV2 Power (W)":260,"Output active power (W)":490,"Output reactive power (kVar)":-0.6,"Grid frequency (Hz)":50.04,"L1 Voltage (V)":231.9,"L1 Current (A)":1.14,"L2 Voltage (V)":234.0,"L2 Current (A)":1.13,"L3 Voltage (V)":231.6,"L3 Current (A)":1.12,"Total production (kWh)":6309,"Total generation time (h)":4532,"Today production (Wh)":330,"Today generation time (min)":72,"Inverter module temperature (oC)":18,"Inverter inner termperature (oC)":34,"Inverter bus voltage (V)":636.6,"PV1 voltage sample by slave CPU (V)":299.3,"PV1 current sample by slave CPU (A)":300.8,"Countdown time (s)":60,"Inverter alert message":0,"Input mode":1,"Communication Board inner message":0,"Insulation of PV1+ to ground":1338,"Insulation of PV2+ to ground":1945,"Insulation of PV- to ground":1570,"Country":"Poland","String 1 voltage (V)":11.4,"String 1 current (A)":23.4,"String 2 voltage (V)":11.3,"String 2 current (A)":23.16,"String 3 voltage (V)":11.2,"String 3 current (A)":0.0,"String 4 voltage (V)":630.9,"String 4 current (A)":0.0,"String 5 voltage (V)":453.2,"String 5 current (A)":0.33,"String 6 voltage (V)":7.2,"String 6 current (A)":0.18,"String 7 voltage (V)":3.4,"String 7 current (A)":63.66,"String 8 voltage (V)":299.3,"String 8 current (A)":30.08}

MichaluxPL, 2021-11-02:
client.tls_set() is beeing run if mqtt_username is not empty, so there is no need to add additional mqtt_ssl config parameter.

MichaluxPL, 2021-11-02:
I've rebuilt the whole MQTT support part of the script. Added option to enable/disable TLS and tested the whole solutoin with Mosquitto. Should work now just fine :)

closed 2021-11-07 by MichaluxPL