MichaluxPL/Sofar_LSW3 #16: Support for inverter HYD3000-ES

jezikk82, 2021-11-11:
Hi,

I tried Your script to communicate with for inverter HYD3000-ES but with no luck. After execution I see following output.  

python3 ./InverterData.py
Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x86soh\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x00\x00(E\xd4\x00\x15')
Received data:  b'\xa5\x10\x00\x10\x15\x00\xb7\x86soh\x02\x010E\x00\x00\x9b\x19\x00\x001\xfe\x8ca\x05\x00\t\x15'
0x0000 - Status: 21
No value in response for register 0x0001
Check register start/end values in config.cfg

python3 ./InverterHWData.py
Modbus request: 0104 2000 000e 7a0e
Hex string to send: A5 1700 1045 0000 86736f68 020000000000000000000000000000 01042000000e7a0e 00 15
Data sent:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x86soh\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x04 \x00\x00\x0ez\x0e\x00\x15')

Data received:  b"\xa5/\x00\x10\x15\x00\xb6\x86soh\x02\x01'E\x00\x00\x92\x19\x00\x001\xfe\x8ca\x01\x04\x1c\x00\x05SM1ES130LBL394V260V100V260X\xb9\x81\x15"
Hex string received: A5 2F 00 10 15 00 B6 86 73 6F 68 02 01 27 45 00 00 92 19 00 00 31 FE 8C 61 01 04 1C 00 05 53 4D 31 45 53 31 33 30 4C 42 4C 33 39 34 56 32 36 30 56 31 30 30 56 32 36 30 58 B9 81 15
Register: 0x2000 , value:0005 ()
Register: 0x2001 , value:534d (SM)
Register: 0x2002 , value:3145 (1E)
Register: 0x2003 , value:5331 (S1)
Register: 0x2004 , value:3330 (30)
Register: 0x2005 , value:4c42 (LB)
Register: 0x2006 , value:4c33 (L3)
Register: 0x2007 , value:3934 (94)
Register: 0x2008 , value:5632 (V2)
Register: 0x2009 , value:3630 (60)
Register: 0x200A , value:5631 (V1)
Register: 0x200B , value:3030 (00)
Register: 0x200C , value:5632 (V2)
Register: 0x200D , value:3630 (60)
{
    "Moc falownika": "15kW",
    "Numer seryjny": "SM1E30LBL394",
    "Wersja oprogramowania": "V260",
    "Wersja sprzętowa": "V100",
    "Wersja DSP": "V260"
}

SN is correct. 
Moc falownika is incorrect (should be 3 kW hybrid). 

Can you help me with this issue ?

MichaluxPL, 2021-11-12:
Hi,

Unfortunately I'm not familiar with RS485 protocol, so I'm not sure what the register addresses are.
Probably, looking at getXXXXXX variables, I would opt for those: 0x00 0x06 0x07 0x0C 0x0D 0x0E 0x0F 0x10 0x11 0x2C 0x12 0x13 0x15 0x18 0x52 0x55 0x1B 0x38 0x39 0x36. So it looks like there are at least 3 different ranges (0x00-0x1B, 0x52-0x55, 0x36-0x38). I would try to configure each one (in config.cfg You can configure just 2 at a time) and try what happens. You can try different ranges, too. Another thing to try is to play with the length parameter in the script (this line: length=binascii.unhexlify('1700') # datalength) - You have to try different values here if You do not get the correct response from the inverter (the one You get is way too short and doesn't contain register values). Wrong length value also produces bad incorrect response.
Also take a look at this page: https://github.com/AndyWhittaker/HYD6000/blob/master/sofar.py 
In this script there are some register's numbers for HYD6000 model - they look different that the one You've found, but maybe this is a good path to follow (I'd also suggest e-mailig Sofar - maybe they will send You MODBUS specification for You inverter model).
To good thing is, You get correct response for the second script, so generally it should work. About wrong inverter's power - this is due to a different value mapping. You'll need the the correct specification for You inverter and change it accordingly in SOFARHWMap.file (Your inverter gives value 5 for the register 0x2000 - clearly for Your inverter type, this is not 15kW).
Hope I've helped just a bit :)
In case of further questions, don't hesitate to ask :)
Michał

jezikk82, 2021-11-13:
Hi,

thx for reply. I already (on 10.11) mailed Sofar for new firmware, ModBus RTU and TCP integration possibilities but so far no answer. 

I'll try to play with the values but can you guide me what value ranger to try here: length=binascii.unhexlify('1700')  ?

Thx.

MichaluxPL, 2021-11-13:
First start with register ranges (try register numbers from the link I've provided). I think here is the main reason You do not get the correct answer. I have another Sofar modbus doc, I've found on elektroda.pl and there register numbers are in range: 0x0200 - 0x0245 (If I were You, I'd give it a shoot :) ). Since this range is quite big, You can try narrow it down first to a smaller set (although it's not a must), or divide it by half and configure as two separate ranges in config.cfg.
Since second script gives You the the correct response, I'll assume the length is OK.
Changing it is the last resort option and actually there is no rule here :), although make minimal changes for the start (i.e by 100, not more).

jezikk82, 2021-11-13:
I'm fighting with the range right now, but I always get  error on the first address. 

No value in response for register 0x0001
No value in response for register 0x0002
No value in response for register 0x0003
...
No value in response for register 0x0201

So manualy I'm trying to adjust the values. Is it possible to change the script to iterate over bigger range without exiting after first empty address ?

MichaluxPL, 2021-11-13:
There is no use of iterating over another values, since lack of it for the first (actually second) one says You don't get the correct answer. Don't look at the first register value - script always says You get here something, but it is not true. Simply by accident the first value, even in a wrong answer, is readable.
In the correct response, You'll a lot of more values, than You get here, so register range is still not correct.
Here is the example correct answer from my inverter:
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00l\x90Pi\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x00\x00(E\xd4\x00\x15')
Received data:  b'\xa5c\x00\x10\x15\x00\xbdl\x90Pi\x02\x01\x08\xcb\x0c8\xf7\x0f\x00\x00\xd8\xc4\x82)\x01\x03P\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x11\xe9\x00\xa5\x03$\x00\x00\x00L\x00\x00\x00E\xff\xbe\x13\x84\t[\x00\x87\tq\x00\x88\tp\x00\x87\x00\x00\x02\xfe\x00\x00\x03\x12\x00;\x00\xf5\x00\x1c\x00,\x19\xa5\x11\xe5\x036\x00<\x00\x00\x00\x01\x00\x00\x05+\t_\x07\x94\x00\x0c\xbd\t\x0b\x15'

MichaluxPL, 2021-11-13:
As far as I can see, looking at a different MODBUS docs, register addresses mostly start from some even number (like 0x0000, 0x0100, 0x1000 etc.) so I'd try sequentially ranges i.e: 0x0000-0x0010, 0x0100-0x0110, 0x0200-0x0210 end so on ...
There is a good chance You'll catch the right starting address eventually :)

jezikk82, 2021-11-16:
Hi again,

I got answer from Sofar with new firmware and new ModBus RTU spec.  Maybe that can help you add support for HYD inverters as well. Thanks.

[SofarHYD ES ME3000SP Modbus protocol.pdf](https://github.com/MichaluxPL/Sofar_LSW3/files/7544510/SofarHYD.ES.ME3000SP.Modbus.protocol.pdf)

MichaluxPL, 2021-11-16:
This doc states the register range should be: 0x0200-0x0255. Try to configure it and test.
If it doesn't work, than there is nothing more I can do - it would require Inverter access and more testing/playing with different values to make it work (although I'm quite sure it eventually would).
You can test one more thing - change command in HW script from 0104 to 0103, set the above register range (or smaller for the start) and test if You get the correct (much longer) answer. If it doesn't, than I'm sorry - back luck :(

closed 2021-12-07 by MichaluxPL