MichaluxPL/Sofar_LSW3 #17: Błąd dekodowania danych

KITT3000, 2021-11-12:
Witam, chciałem użyć tego skryptu do wysyłania danych do PVmonitor.pl, ale niestety nie może on zdekodować danych, aby zostały wyświetlone jak w przykładzie, tutaj wpis z konsoli:
```
Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00AW\x96h\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x00\x00(E\xd4\x00\x15')
Received data:  b'\xa5c\x00\x10\x15\x00\x82AW\x96h\x02\x01\x08\x17\x94\x00\x90!\x00\x00\xb3\x14\xfa`\x01\x03P\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0e\xb7\x00C\x03#\x00\x00\x00\x19\x00\x00\x00\x15\xff\xbf\x13\x8a\t\x88\x00_\tv\x00_\t\x95\x00a\x00\x00\x0c"\x00\x00\n0\x00\x12\x00\x86\x00\x0e\x00\x1f\x19\xfa\x0e\xba\x03*\x00<\x00\x00\x00\x01\x00\x00\x04\xeb\x08.\x06\xab\x00\x0c\xff\xb5\n\x15'
Traceback (most recent call last):
  File "F:\Pobrane\Przeglądarka\Sofar_LSW3-1.5\InverterData.py", line 169, in <module>
    parameters=json.loads(txtfile.read())
  File "C:\Users\xXx\AppData\Local\Programs\Python\Python39\lib\encodings\cp1250.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 7950: character maps to <undefined>
```

KITT3000, 2021-11-12:
OK, znalazłem rozwiązanie w zamkniętych problemach (#6)

closed 2021-11-12 by KITT3000

MichaluxPL, 2021-11-12:
Z komunikatu wynika, że jest problem z kodowaniem znaków w pliku
SOFARMap.xml (jak widzę jedziesz spod Windowsa ;)).
Zerknij na wątek na elektrodzie:
https://www.elektroda.pl/rtvforum/topic3698233-90.html
W którymś miejscu, ktoś napisał jak przerobić kodowanie :)
Albo zmień w skrypcie linię, gdzie otwierany jest plik SOFARMap.xml na:  with
open("./SOFARMap.xml", encoding="utf8") as txtfile:

Michał