MichaluxPL/Sofar_LSW3 #18: Integration with Domoticz

adamb94, 2021-11-22:
Hello,

Do you know how to integrate that script with Domoticz? I see input in Mosuitto but I don't know how to add it to Domoticz

MichaluxPL, 2021-11-22:
Current MQTT support does not include Domoticz integration. AFAIK Domoticz requires device ID to be sent together with actual data - in order to know which device particular data is for. Probably it wouldn't be that much effort to add such a functionality - I'd suggest adding additional DomoticzID parameter in SofarMap.xml and use it in the script to send it's value to MQTT together with inverter's data.
Other way I can see is to write simple bash script, that will parse Sofar script output and send data to Domoticz using curl and Domoticz's API interface. That should be pretty straight forward :)

MichaluxPL, 2021-11-23:
So .... since I'm using Domitocz myself, I've decided to add support for it to the script :)
Look at version 1.6 - You'll find it there.
Have fun :)

adamb94, 2021-11-23:
Hello,
Many thanks it's working perfectly :)
I've only had issue with Status(set as text).
I needed to edit line 201:
Changed from:
response='"'+option["valuePL"]+'"'
to:
response=option["valuePL"]

MichaluxPL, 2021-11-23:
Glad I could help :)
Text values are kind of not suitable for Domoticz, I guess.
If you change this one, script will send data to Domoticz, but it will crash at the end trying to print JSON output.
I've fixed it in much more elegant way - look at v1.61 (it'll appear in a moment)

adamb94, 2021-11-23:
You are the man :)
I'm not good developer. I have only basic.
Now everything is working perfectly!

Dzięki wielkie i masz piwo u mnie :)

closed 2021-11-23 by adamb94

MichaluxPL, 2021-11-23:
Nie ma problemu :)