MichaluxPL/Sofar_LSW3 #20: LSE-3 logger - No value in response for register 0x0000

wardwygaerts, 2021-12-17:
Hi

I tried to connect to my [LSE-3 logger (eth type)](https://www.sofarsolar.com/product-detail/499/Stick-Logger),but I get these message:
```
No value in response for register 0x0000
Check register start/end values in config.cfg
```

But I have no clue about these register start/end values

Device serial number 210xxxxxxx
Firmware version ME_0C_270A_1.05

wardwygaerts, 2021-12-19:
I swapped my eth logger by the LSW3 version (firmware LSW3_15_270A_1.32)

First I had some issue with opening the *SOFARMap.xml* file, but this is solved now.
But also here I get lots of errors regarding `invalid literal for int() with base 16: '  '`

I printed the hexval which is used in `twosComplement_hexof` and this is most of the times empty, so something isn't not correct in this command: `str(''.join(hex(ord(chr(x)))[2:].zfill(2) for x in bytearray(data))+'  '+re.sub('[^\x20-\x7f]', '', ''))[p1:p2], hexpos`
I also removed the sys.exit in this function.

```
hex -0003-
hex -8b15-
hex -  -
invalid literal for int() with base 16: '  '
No value in response for register 0x0003
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0004
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0005
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0006
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0007
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0008
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0009
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x000A
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x000B
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x000C
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x000D
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x000E
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x000F
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0010
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0011
[... 123 lines cut ...]
No value in response for register 0x010F
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0110
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0111
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0112
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0113
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
No value in response for register 0x0114
Check register start/end values in config.cfg
hex --
invalid literal for int() with base 16: ''
```

MichaluxPL, 2021-12-20:
You probably do not get valid response from the logger/inverter - that's why the script complains about wrong data.
Please send the response string You get (set verbose=1 in config file).

wardwygaerts, 2021-12-20:
These are the lines I get back with verbose on
```
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x1ae\xe2i\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x00\x00(E\xd4\x00\x15')
Received data:  b'\xa5\x13\x00\x10\x15\x00=\x1ae\xe2i\x02\x01(\xb9\x0c\x00o\x0f\x00\x00\x92\xc3\xb3a\x01\x90\x02\x00\x03\xac\x15'
```

MichaluxPL, 2021-12-21:
As I've suspected, You don't receive a valid response from inverter. Received data does not contain actual performance values (it is way too short). This is most likely due to incorrect register's address configuration (in config.cfg). I'd suggest contacting Your inverter manufacturer for MODBUS specification - there You should find the correct register address ranges to configure (both for performance and HW data). You can also look for it in Internet.
Another approach is to play around with different register ranges - maybe You'll catch the right one eventually :)

mcbeth111, 2022-05-28:
Hi, I have a similar problem. My LSE-3 USB (ME_0D_270A_1.08) responds to solarman frame very strange:
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\myloggersn\x00\x00\x00\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x02\x10$\x00\x10=\r\x00\x15')
 a5 17 00 10 45 00 00 myloggersn 00 00 00 00 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 02 10 24 00 10 3d 0d 7b 15
Received data:  b'\xa5\x17\x00\x10E\x03\x00\x8a\x02'
 a5 17 00 10 45 03 00 8a 02 - there should be 16 words of data but even frame header looks bad

Just for experiment:
When I send just '00' i receive '00 0a 00 10 17 03 16 8a 02'
When  00 11 22 33 44 55 66 77 88 99 response is  00 11 22 33 44 03 66 f7 02
-funny isn't it ?

I can switch logger to 'transparent' via config_hide.html. In this mode it accepts pure modbus and responds with correct data (unfortunately in transparent mode cloud data logging doesn't work). 
A verified correct packet wrapped in solarman v5 frame (in standard 'data collection' logger mode) returns junk (?)

Any ideas ?

MichaluxPL, 2022-05-31:
Hi,

Sorry, I can't help :(
Maybe Your logger uses a different frame format or even the communication
protocol has changed (Your logger's version differs from mine).

Michal

ilexpl01, 2022-10-01:
> Hi, I have a similar problem. My LSE-3 USB (ME_0D_270A_1.08) responds to solarman frame very strange: Sent data: [...]

Hi, 

have You managed to solve the problem ?

Got the same issue once I have replaced LSW-3 to LSE-3.

Frankstar, 2024-08-01:
> Hi, I have a similar problem. My LSE-3 USB (ME_0D_270A_1.08) responds to solarman frame very strange: Sent data: [...]

same problems.
but in transparency mode i get no response data at all ....