MichaluxPL/Sofar_LSW3 #25: FreeBSD port and wrong degree symbol

mzary, 2022-01-30:
Thanks for this app, it's really appreciated since protocol v5 was a real blackbox until then. 

There is one issue I have to report. The symbol used in xml file for temperature is probably wrong, it is now "º" U+00BA (masculine ordinal indicator), but this should be "°" U+00B0.

I am using this under FreeBSD. If there is real interest and more people are willing to run it under FreeBSD, will eventually create a port.

MichaluxPL, 2022-01-30:
Under Linux it works just fine, but feel free to change it as You require :)

closed 2022-01-30 by MichaluxPL

mzary, 2022-01-30:
Thanks, it was a minor, cosmetical issue that had nothing to do with the operating system, but it was a bit confusing.