MichaluxPL/Sofar_LSW3 #27: Domoticz-aktualizacja danymi z inwertera

ozogp, 2022-02-07:
Witam, szukam przyczyny conajmniej dziwnego zachowywania pozyskanych danych lub ich braku w Domoticz. 
Instancja Domoticz 2021.1 (build 13949) (Synology-Docker) - chcę pozyskać 12 danych z inwertera (Moc PV1,Moc PV2, Napięcie PV1, Napięcie PV2, Prąd PV, Prąd PV2, Dzienna produkcja, Całkowita produkcja, Moc czynna, Napięcie L1, Napięcie L2, Napięcie L3, Temperatura falownika, Temperatura modułu. 
Wszystkie potrzebne virtualne urządzenia w Domoticz utworzone, skrypty poprawnie odpowiadają i wyświetlają się kompletne wyniki na terminalu - wszystko wygląda ok. Jednak raz tylko część danych pojawia się w Domoticz (np. 3 lub 2) , raz pojawiają się wszystkie które zostały skonfigurowane (12). Nie jest to regularne - zazwyczaj zawsze jest Napięcie PV1. Skrypt działa przez crone co 2 min. zmiana na inny czas np 10 min. też nie dała rezultatu. 
Może jakiś pomysł?

MichaluxPL, 2022-02-09:
Zacząłbym od podejrzenia na poziomie serwera MQTT, co do niego przychodzi - czy za każdym razem wpadają informacje o wszytskich parametrach ? Jeśli nie, to będę musiał się temu przyjrzeć :)

ozogp, 2022-02-10:
Dzień dobry, 
za radą ustawiłem klienta MQTT do nasłuchiwania komunikacji z Inwertera do Domoticza.
Komunikacja jak najbardziej odbywa się co 2 minuty ale wartości są pojedyńcze. 
mqtt client: https://ibb.co/5rN2r5C
log: https://ibb.co/CvvdBWK
Genaralnie jak widać w ramach jednej sesji wysyła np. 1 lub 2 komunikaty o stanie. 
Patrząć od strony samego Domoticza z terminala: https://ibb.co/6DBRW4z wszystko wygląda OK, dane pobierają się prawidłowo. 
Generalnie nie mam czego czepić.

MichaluxPL, 2022-02-10:
OK, dzięki za info. Postaram się zajrzeć do tego w niedzielę :)

ozogp, 2022-02-11:
Dziekuję, 
widok Domoticza z dzisiaj: https://ibb.co/LvXyG6c

MichaluxPL, 2022-02-13:
Przeprowadziłem dzisiaj szybkie testy integracji z Domoticzem via MQTT i u mnie zachowuje się to poprawnie.
Wszystkie komunikaty wysyłane przez skrypt docierają do MQTT, a co za tym idzie do Domoticza. Ja korzystam z Mosquitto.
Włącz verbose=1 w configu i podeślij zrzut ekranu z output'u skryptu - tam powinna być informacja co dokładnie skrypt próbuje wysłać do MQTT. Podeślij też zrzut z tego, co dociera do MQTT (tak, aby było widać dokładnie wszystkie komunikaty - wraz z ich zawartością - np. jakiś zrzut z konsoli, a nie GUI).

ozogp, 2022-02-15:
Witam,
również mam Mosquitto 
LOG MQTT: https://ibb.co/RYRGmY7
TERMINAL DOMOTICZA: https://ibb.co/vYDFJkZ , https://ibb.co/fQ57CSj , https://ibb.co/LvPM5P1
DOMOTICZ WWW: https://ibb.co/8z8fQt8

MichaluxPL, 2022-02-15:
Z outputu skryptu wynika, że wysyła wszystkie komunikaty do MQTT.
Podeślij listę komunikatów MQTT, ale nie z logu MQTT tylko np. z polecenia (podstaw swoje wartości pod IP, usera i hasło :) ):
mosquitto_sub -v -h MOSQUITTO_IP -p 1883 -u USER -P 'HASŁO' -t 'domoticz/in/#'
Najpierw odpal powyższą komendę, potem w innym terminalu skrypt monitorujący. Podeślij to, co się pojawi jako output z powyższej komendy po zakończeniu pracy skryptu.

ozogp, 2022-02-15:
https://ibb.co/rcf83dV

https://ibb.co/6WCvTLm

https://ibb.co/njWVJZz

MichaluxPL, 2022-02-15:
Ok, już coś więcej widać, ale nie przesłałeś zrzutów z output samego skryptu, aby skorelować jedno z drugim.
Odpal jeszcze raz (albo kilka razy) i wrzuć zrzuty ze skryptu i MQTT (ze skryptu tylko wycinki z info o przesłani do MQTT).

ozogp, 2022-02-15:
Wykonanie skryptu
TERMINAL DOMOTICZA: https://ibb.co/3vLY1YT , https://ibb.co/vjbymd8 , https://ibb.co/jTFNgB3 , https://ibb.co/YhdXXMW
MQTT: https://ibb.co/xfHwwSF
DOMOTICZ: https://ibb.co/VL1fCsj

MichaluxPL, 2022-02-15:
W drugim przypadku do MQTT dotarło 9  wiadomości - tak gwoli szczegółu. Co nie zmienia faktu, iż nie docierają wszystkie wysyłane, a tych jak widać powinno obyć, jeśli dobrze liczę, 17.
Rzeczywiście wygląda jakby było zrywane połączenie. Zastanawiam się czy w grę tu nie wchodzi fakt, iż Mosquitto jest w kontenerze. Ja mam postawione normalnie na OSie i nie obserwuję takich objawów.
Ale jeszcze popatrzę.

MichaluxPL, 2022-02-15:
Dobra, już chyba wiem w czym problem :)
Postaram się wrzucić poprawkę niebawem.

ozogp, 2022-02-15:
> W drugim przypadku do MQTT dotarło 9 wiadomości - tak gwoli szczegółu. Co nie zmienia faktu, iż nie docierają wszystkie wysyłane, a tych jak widać powinno obyć, jeśli dobrze liczę, 17. Rzeczywiście [...]

ozogp, 2022-02-15:
> 
Byłoby świetnie, na podstawie odczytów z Invertera chciałbym wykonać automatykę w kontekście: jeżeli napięcie na fazie L1,L2,L3 jest za wysokie to włącz grzałkę na zasobniku wody, klimę itd.

ozogp, 2022-02-15:
> Byłoby świetnie, na podstawie stabilnych i powtarzalnych odczytów z Invertera chciałbym wykonać automatykę w kontekście: jeżeli napięcie na fazie L1,L2,L3 jest za wysokie to włącz grzałkę na zasobniku wody, klimę itd.

Byłoby świetnie, na podstawie stabilnych i powtarzalnych odczytów z Invertera chciałbym wykonać automatykę w kontekście: jeżeli napięcie na fazie L1,L2,L3 jest za wysokie to włącz grzałkę na zasobniku wody, klimę itd.

ozogp, 2022-02-15:
> Dobra, już chyba wiem w czym problem :) Postaram się wrzucić poprawkę niebawem.

Byłoby świetnie, na podstawie stabilnych i powtarzalnych odczytów z Invertera chciałbym wykonać automatykę w kontekście: jeżeli napięcie na fazie L1,L2,L3 jest za wysokie to włącz grzałkę na zasobniku wody, klimę itd.

MichaluxPL, 2022-02-15:
Poprawka do skryptu wpadła do GITa :) Zassaj, przetestuj i daj znać.

ozogp, 2022-02-15:
> Poprawka do skryptu wpadła do GITa :) Zassaj, przetestuj i daj znać.

Czy mogę przegrać stare pliki config.cfg, SOFARMap.xml, SOFARHWMap.xml?

MichaluxPL, 2022-02-15:
Tak, do podmiany jest tylko sam skrypt.

ozogp, 2022-02-15:
Trzy iteracje skryptu MQTT: https://ibb.co/jHrVMXf

Patrząc od strony terminala Domoticza wykonywanie samego skryptu się zwolniło ale przechodzi dokładnie jak te wcześniejsze więc nie pisałem.

MichaluxPL, 2022-02-15:
A jak to wygląda w logu MQTT ? Nadal jest info o zerwaniu połączenia przed publikacją wszystkich komunikatów ?

ozogp, 2022-02-15:
LOG MQTT: https://ibb.co/KN161yf

MichaluxPL, 2022-02-15:
Wrzuciłem kolejny fix - tym razem wszystkie komunikaty są wysyłane jako jedna komenda, więc nie powinno mieć miejsca przerwanie publikacji w trakcie. Jeśli to nie zadziała, to na razie nie mam więcej pomysłów.
Do podmiany tylko sam skrypt.

ozogp, 2022-02-16:
[1.61] - działa - z różnym skutkiem, ale działa 
[1.62] - nie działa - nie ma żadnych informacji w Domoticz i MQTT nie otrzymuje żadnych informacji

MichaluxPL, 2022-02-16:
Wrzuciłeś skrypt, który dzisiaj zacommitowałem ? (1.63)

ozogp, 2022-02-16:
Przepraszam ale nie wiem widzę tylko wersję ostatnią 1.62 https://ibb.co/wKYDKpN

MichaluxPL, 2022-02-16:
Głównej wersji nie zmieniałem na razie.
Zrób normalne pull'a i zobacz jaką wersję masz w skrypcie wpisaną.

ozogp, 2022-02-16:
Mea culpa. 
czy możesz wstawić wersję skryptu 1.62, cały czas korzystałem 1.61 - przepraszam.

MichaluxPL, 2022-02-16:
Zaciągnij to, co jest w GIT.

MichaluxPL, 2022-02-16:
Dobra, zrobiłem Ci wersję 1.64.

ozogp, 2022-02-21:
Dzień dobry, 
przepraszam, że dopiero teraz ale ferie :) Podegrałem wersję 1.64 skryptu, zweryfikowałem aby mieć pewność. Skrypt się wykonyje na terminalu jest OK, ale nic nie wpada do Domoticza, ani nic nie ma w logach MQTT.
Czy możemy wrócić do wersji 1.62 - obecnie mi działa wersja skryptu 1.61. Przepraszam za zamieszanie

MichaluxPL, 2022-02-21:
Wrzuciłem wersję 1.65 - przetestuj to.

ozogp, 2022-02-21:
Pobrałem skrypt, wgrałem go na PRD. 
Wykonałem ręcznie 9:48 - pobrał wszystkie wartości i wpisał do Domoticz. 
o 9:50 wykonał się skrypt z crone - pobrał i wpisał wszystkie wartości do Domoticza. 

Poczekam do 10:00 - będzie kolejne wykonanie skryptu z crone . Napisze jaki wynik.

MichaluxPL, 2022-02-21:
Na zdrowie :)

ozogp, 2022-02-21:
o 10:10 wykonał się skrypt z crone - pobrał i wpisał wszystkie wartości do Domoticza.

Bardzo dziekuję, atakuję teraz automatyzację. 
Napięcia na inverterze zbliżają się pod 250 V muszę w takich przypadkach pozałączać grzałkę na bojlerze, klimę, do się wyłączy

ozogp, 2022-02-21:
Jest stabilnie, można zamykać :)

closed 2022-02-21 by ozogp

MichaluxPL, 2022-02-21:
Problem polegał na tym, iż połączenie do MQTT było zamykane szybciej niż schodziły komunikaty. Wymagane było zastosowanie dodatkowej procedury loop i kilku innych patentów.
Co do wersji w skrypcie, to zostanie poprawiona :)