MichaluxPL/Sofar_LSW3 #30: UnicodeDecodeError: 'charmap' codec can't decode

MartinJSa, 2022-02-28:
Hi, firstly thanks very much for your hard work on this, very keen to get it working on my ME3000SP. I'm using Python in Windows 11

When I run the HWData script everything works fine, but when I run InverterData.py I get:

```
PS C:\Users\marti\OneDrive\Documents\Sofar_LSW3-main-v1-8\Sofar_LSW3-main> python3 .\InverterData.py
Connecting to logger... connected successfully !
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00?\xf1\x13k\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x00\x00(E\xd4\x00\x15')
Received data:  b'\xa5\x10\x00\x10\x15\x00/?\xf1\x13k\x02\x01w\x88L7M\r\x00\x00;p\xd0*\x05\x00\x9b\x15'
Traceback (most recent call last):
  File "C:\Users\marti\OneDrive\Documents\Sofar_LSW3-main-v1-8\Sofar_LSW3-main\InverterData.py", line 186, in <module>
    parameters=json.loads(txtfile.read())
  File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.9_3.9.2800.0_x64__qbz5n2kfra8p0\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x81 in position 7995: character maps to <undefined>
```

I presume there is a character map issue going on here? Any suggestions please?

MichaluxPL, 2022-02-28:
Open SOFARMap.xml in notepad (or other text editor) and remove/change all Polish characters, or those that does not look OK. That should do it :)
BTW: actually error message tells which character it has problem with: "codec can't decode byte 0x81 in position 7995" - looking at SOFARMap.xml file most probably it complains about this one: BŁĄD :)

MartinJSa, 2022-02-28:
I think I may have made a bit of progress, I changed line 185 to be the following:

`with open("./SOFARMap.xml", encoding="utf-8") as txtfile:`

I now get:
```
PS C:\Users\marti\OneDrive\Documents\Sofar_LSW3-main-v1-8\Sofar_LSW3-main> python3 .\InverterData.py
Connecting to logger... connected successfully !
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00?\xf1\x13k\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x00\x00(E\xd4\x00\x15')
Received data:  b'\xa5\x10\x00\x10\x15\x00L?\xf1\x13k\x02\x01\xc8\x91L7\x9f\x16\x00\x00;p\xd0*\x05\x00m\x15'
0x0000 - Inverter status: 21
No value in response for register 0x0001
Check register start/end values in config.cfg
```

MichaluxPL, 2022-02-28:
Glad You've made it :)
Now, most probably, You have incorrect register ranges configured in config.cfg.
You have two options:
1. Find MODBUS registers specification for You inverter and use it to define the correct register's range.
2. Play with different register ranges to find the correct one.
Unfortunately I wont be able to help You with either one :(

MartinJSa, 2022-02-28:
Thanks for the quick replies! 

What's confusing me now is 0x0001 is a valid register, it seems to be a fault register:

[image]

If I skip that section and start at 0x0006 I get: 

```
PS C:\Users\marti\OneDrive\Documents\Sofar_LSW3-main-v1-8\Sofar_LSW3-main> python3 .\InverterData.py
Connecting to logger... connected successfully !
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00?\xf1\x13k\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x06\x00"%\xd2\x00\x15')
Received data:  b'\xa5\x10\x00\x10\x15\x00]?\xf1\x13k\x02\x01\xa7\x95L7~\x1a\x00\x00;p\xd0*\x05\x00D\x15'
0x0006 - PV1 Voltage: 2.1V
No value in response for register 0x0007
Check register start/end values in config.cfg
```

I guess my inverter is returning empty values when the script is expecting every register to be populated?

[Registry_Details.pdf](https://github.com/MichaluxPL/Sofar_LSW3/files/8155347/Registry_Details.pdf)

MichaluxPL, 2022-02-28:
Currently You don't get any valid info from the inverter - it is simply too short (does not contain any inverter's data - only some error code probably). So don't look for 0x0006 value since it totally misleading now (it's just a coincidence it decodes to some value).
Check if this is a correct MODBUS specification for Your type of inverter (as far as I can see You have a hybrid one, while attached documentation seams to be for a normal one - like mine: K-TLX).
So first try to play with register ranges to get much longer "Received data" byte string. As long as You get such a short one, it means You don't get any valid production data.

MichaluxPL, 2022-02-28:
For a correct response example string look at closed issue #17: https://github.com/MichaluxPL/Sofar_LSW3/issues/17 (it's in polish, but first message contains a correct response byte string).

MartinJSa, 2022-03-01:
Ah great, thanks very much I'll do more digging!

MartinJSa, 2022-03-04:
Just following up on this, I've had this link passed to me which looks more promising as MODBUS spec for the ME3000SP. https://github.com/greentangerine/ME3000/blob/master/SOFARSOLAR%20ModBus-RTU%20Communication%20Protocol.pdf

I'm trying out a few register ranges, and depending on what I go for, I'm getting this error:

```
Connecting to logger... connected successfully !
Traceback (most recent call last):
  File "C:\Users\marti\OneDrive\Documents\Sofar_LSW3-main-v1-8\Sofar_LSW3-main\InverterData.py", line 141, in <module>
    crc=binascii.unhexlify(str(hex(libscrc.modbus(businessfield))[4:6])+str(hex(libscrc.modbus(businessfield))[2:4])) # CRC16modbus
binascii.Error: Odd-length string
```
Could you offer any ideas on what's going wrong here please?

MichaluxPL, 2022-03-04:
Most likely You have a wrong register's range configured - try to increase/decrease register's range by 1.

closed 2022-03-14 by MichaluxPL