MichaluxPL/Sofar_LSW3 #32: Błędne dane z loggera LSW-3  ver. LSW3_15_270A_1.32

paku2020, 2022-03-11:
Witam.
Udało się uruchomić integrację.
Jednak jest problem który polega na dziwnych danych.
Wygląda to tak:
```
user@debian~/SofarSolar/Sofar_LSW3$ python3 InverterData.py
Connecting to logger... connected successfully !
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00F4\x17j\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x00\x00(E\xd4\x00\x15')
Received data:  b"\xa5c\x00\x10\x15\x00uF4\x17j\x02\x01\x93\xf1\x03\x00\x8a\x08\x00\x00\x9f\x17'b\x01\x03P\x00\x03\xff\xc0\x05\x81\xff\xff\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00ly\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00<\x00\x0b\x00\x00\x00\x0b\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x19\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xc7\x00\x00\xe4v`\x15"
0x0000 - Status: "BŁĄD"
0x0001 - Kod błędu 1: -64
0x0002 - Kod błędu 2: 1409
0x0003 - Kod błędu 3: -1
0x0004 - Kod błędu 4: "ID50 Duża różnica w poziomie próbkowania częstotliwości pomiędzy nadrzędnym i podrzędnym DSP"
0x0005 - Kod błędu 5: "Brak błędów"
0x0006 - Napięcie PV1: 0.0V
0x0007 - Prąd PV1: 0.0A
0x0008 - Napięcie PV2: 0.0V
0x0009 - Prąd PV2: 0.0A
0x000A - Moc PV1: 0W
0x000B - Moc PV2: 0W
0x000C - Moc czynna: 0W
0x000D - Moc bierna: 0.0kVar
0x000E - Częstotliwość sieci: 0.0Hz
0x000F - Napięcie L1: 0.0V
0x0010 - Prąd L1: 277.69A
0x0011 - Napięcie L2: 0.0V
0x0012 - Prąd L2: 0.0A
0x0013 - Napięcie L3: 0.0V
0x0014 - Prąd L3: 0.0A
0x0016 - Całkowita produkcja: 0kWh
0x0018 - Całkowity czas pracy: 11h
0x0019 - Dzienna produkcja: 0Wh
0x001A - Dzisiejszy czas pracy: 11min
0x001B - Temperatura modułu: 0ºC
0x001C - Temperatura falownika: 0ºC
0x001D - Napięcie szyny falownika: 0.0V
0x001E - Napięcie PV1 próbkowane przez podrzędny CPU: 0.0V
0x001F - Prąd PV1 próbkowany przez podrzędny CPU: 0.0A
0x0020 - Czas próbkowania: 25s
0x0021 - Wiadomość alarmowa: 0
0x0022 - Tryb wejściowy: 0
0x0023 - Wewnętrzny komunikat szyny: 0
0x0024 - Izolacja PV1+ względem ziemi: 0
0x0025 - Izolacja PV2+ względem ziemi: 0
0x0026 - Izolacja PV- względem ziemi: 199
0x0027 - Kraj: "Niemcy"
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00F4\x17j\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x10\x00\x10E\xc3\x00\x15')
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00F4\x17j\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x10\x00\x10E\xc3\x00\x15')
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00F4\x17j\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x10\x00\x10E\xc3\x00\x15')
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00F4\x17j\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x10\x00\x10E\xc3\x00\x15')
^CTraceback (most recent call last):
  File "/home/kuma/SofarSolar/Sofar_LSW3/InverterData.py", line 164, in <module>
    data = clientSocket.recv(1024);
KeyboardInterrupt

user@debian:~/SofarSolar/Sofar_LSW3$
```

Na falowniku jest wszystko dobrze, brak błędów i praca normalna.
Inni z tego typu loggerem nie mają tego problemów.
W czym może być problem? Może wersja oprogramowania na loggerze? 
Zestaw był zakupiony w luty tego roku.

MichaluxPL, 2022-03-11:
Prawdopodobnie drugi podany zakres rejestrów (w config.cfg) jest nieprawidłowy, gdyż nie jest zwracana żadna odpowiedź z loggier'a/falownika.
Proponuję pokombinować z zakresem.
Ewentualnie w ramach testu można drugi zakres wpisać jako pierwszy i zobaczyć, czy coś jest zwracane.

paku2020, 2022-03-11:
Zamieniłem zakres drugi z pierwszym i odwrotnie.
Mam taki wynik

```
user@debian:~/SofarSolar/Sofar_LSW3$ python3 InverterData.py
Connecting to logger... connected successfully !
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00F4\x17j\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x10\x00\x10E\xc3\x00\x15')
Received data:  b"\xa53\x00\x10\x15\x00\xbfF4\x17j\x02\x01\xb7\x07\x04\x00\xae\x1e\x00\x00\x9f\x17'b\x01\x03 \x00\x03\xff\xc0\x05\x81\xff\xff\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00ly3\x15"
0x0105 - Napięcie string 1: 0.3V
0x0106 - Prąd string 1: -0.64A
0x0107 - Napięcie string 2: 140.9V
0x0108 - Prąd string 2: -0.01A
0x0109 - Napięcie string 3: 0.2V
0x010A - Prąd string 3: 0.0A
0x010B - Napięcie string 4: 0.0V
0x010C - Prąd string 4: 0.0A
0x010D - Napięcie string 5: 0.0V
0x010E - Prąd string 5: 0.0A
0x010F - Napięcie string 6: 0.0V
0x0110 - Prąd string 6: 0.0A
0x0111 - Napięcie string 7: 0.0V
0x0112 - Prąd string 7: 0.0A
0x0113 - Napięcie string 8: 0.0V
0x0114 - Prąd string 8: 0.0A
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00F4\x17j\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x00\x00(E\xd4\x00\x15')
Received data:  b"\xa5c\x00\x10\x15\x00\xc0F4\x17j\x02\x01\xb7\x07\x04\x00\xae\x1e\x00\x00\x9f\x17'b\x01\x03P\x00\x03\xff\xc0\x05\x81\xff\xff\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00ly\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00<\x00\x10\x00\x00\x00\x12\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x19\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01&\x00\x00z\xcav\x15"
0x0000 - Status: "BŁĄD"
0x0001 - Kod błędu 1: -64
0x0002 - Kod błędu 2: 1409
0x0003 - Kod błędu 3: -1
0x0004 - Kod błędu 4: "ID50 Duża różnica w poziomie próbkowania częstotliwości pomiędzy nadrzędnym i podrzędnym DSP"
0x0005 - Kod błędu 5: "Brak błędów"
0x0006 - Napięcie PV1: 0.0V
0x0007 - Prąd PV1: 0.0A
0x0008 - Napięcie PV2: 0.0V
0x0009 - Prąd PV2: 0.0A
0x000A - Moc PV1: 0W
0x000B - Moc PV2: 0W
0x000C - Moc czynna: 0W
0x000D - Moc bierna: 0.0kVar
0x000E - Częstotliwość sieci: 0.0Hz
0x000F - Napięcie L1: 0.0V
0x0010 - Prąd L1: 277.69A
0x0011 - Napięcie L2: 0.0V
0x0012 - Prąd L2: 0.0A
0x0013 - Napięcie L3: 0.0V
0x0014 - Prąd L3: 0.0A
0x0016 - Całkowita produkcja: 0kWh
0x0018 - Całkowity czas pracy: 16h
0x0019 - Dzienna produkcja: 0Wh
0x001A - Dzisiejszy czas pracy: 18min
0x001B - Temperatura modułu: 0ºC
0x001C - Temperatura falownika: 0ºC
0x001D - Napięcie szyny falownika: 0.0V
0x001E - Napięcie PV1 próbkowane przez podrzędny CPU: 0.0V
0x001F - Prąd PV1 próbkowany przez podrzędny CPU: 0.0A
0x0020 - Czas próbkowania: 25s
0x0021 - Wiadomość alarmowa: 0
0x0022 - Tryb wejściowy: 0
0x0023 - Wewnętrzny komunikat szyny: 0
0x0024 - Izolacja PV1+ względem ziemi: 0
0x0025 - Izolacja PV2+ względem ziemi: 0
0x0026 - Izolacja PV- względem ziemi: 294
0x0027 - Kraj: "Niemcy"
user@debian:~/SofarSolar/Sofar_LSW3$
```
Jednak jak widać pobrane dane są dość dziwne i całkowicie odbiegają od rzeczywistości.
Jak mogę pokombinować z zakresami ?

MichaluxPL, 2022-03-11:
Jak widać komunikacja przebiega bez problemu - to akurat dobrze :)
Natomiast ewidentnie albo podawane zakresy rejestrów są nie prawidłowe dla tego typu falownika, albo pod poszczególnymi numerami rejestrów są przechowywane inne wartości niż to zostało zdefiniowane w SOFARMap.xml.
Najlepiej zdobyć dokumentację MODBUS dla dokładnie tego modelu falownika i z niej pobrać wymagane zakresy oraz co każdy z rejestrów przechowuje.
Konieczna może się także okazać zmiana wpisów w SOFARMap.xml - tam poszczególny numery rejestrów są przypisane do odpowiednich rodzajów wartości. Wszystko to trzeba zrobić na podstawie dokumentacji - tam powinno być napisane, który numer rejestru przechowuje jaką wartość.

paku2020, 2022-03-11:
Oki, to pisze do Sofara o dokumentację modbus.
Chodzi  o dokumentację do loggera, czy do falownika ? Żebym wiedział co i jak mam im opisać :)
To ma być dokumentacja modbus dla falownika o takim i takim oznaczeniu modelu i sn i najlepiej chyba jeszcze podać wersje oprogramowania ?

MichaluxPL, 2022-03-11:
Dokumentacja dla falownika, bo to jego rejestry odczytuje logger.
Podać wersji softu pewnie nie zawadzi :)

paku2020, 2022-03-11:
Oki, już wysłałem do nich zapytanie, podałem wszystkie dane jakie tylko mam :)
Wydaje mi się że musieli coś zmienić w oprogramowaniu bo dziwne że u innych działa a u mnie nie :)
Jak dostanę odpowiedz to dam znać i może coś podziałamy

paku2020, 2022-03-11:
[SOFAR-G3 External Modbus Protocol-Engilsh-V211103.xlsx](https://github.com/MichaluxPL/Sofar_LSW3/files/8232395/SOFAR-G3.External.Modbus.Protocol-Engilsh-V211103.xlsx)

Dostałem odpowiedz od sofara, ale niestety jestem za słaby w tym żeby wiedzieć jak tego użyć.
Mogę liczyć na małą pomoc ?

MichaluxPL, 2022-03-12:
W pliku masz wszystkie potrzebne dane. Poszczególne kolumny oznaczają:
Register address -> adres rejestru przechowującego wartość dla danego parametru (czasem jeden parametr ma wartość w dwóch rejestrach lub więcej). Na podstawie tych wartości budujesz zakres, który chcesz odczytywać. Musisz go wpisać w config.cfg (możesz zdefiniować max 2 takie zakresy obecnie).
Filed -> informacja, co w danym rejestrze jest przechowywane (poszczególne parametry robocze falownika)
Type -> typ wartości, np. U64 to pewnie unsigned integer o rozmiarze 64bity, czyli max wartość to 2 do potęgi 64
Accuracy -> dokładność przechowywanej wartości (np 1, 0.1 itd.)
Unit -> jednostka (np. kWh)
Min/Max -> jaką minimalną i maksymalną wartość może dana zmienna przyjąć
Read/Write -> czy wartość jest tylko do odczytu, czy też do zapisu (np. w celu sterowania falownikiem)
Remarks -> opis/uwagi itp.
User -> mówi dla kogo dany parametr jest przeznaczony do wykorzystania (użytkownik/instalator).

Zatem musisz wybrać zakres rejestrów, który chcesz czytać (max 2 zakresy) i wpisać je w config.cfg. Następnie dokładnie te numery rejestrów zdefiniować w SOFARMap.xml - podajesz numer rejestru oraz info co przechowuje, w jakich jednostkach, oraz z jaką dokładnością (stąd skrypt wie co jest gdzie i poprawnie wyświetla nazwy poszczególny parametrów)
I to powinno załatwić sprawę. Przy czym jeszcze będziesz musiał obsłużyć dodatkowy myk - niektóre parametry są przechowywane w więcej niż 1 rejestrze. Obecnie skrypt ma zdefiniowane dwie takie wartości, odczytywane z rejestrów 0x0015/16 (całkowita produkcja) i 17/18 (całkowity czas generacji). U Ciebie może to wyglądać inaczej, więc musisz to wziąć pod uwagę (np. zmienić w skrypcie te adresy, na takie, pod którymi Twój falownik przetrzymuje obie te wartości).

Z tego co widzę, Twój falownik przechowuje swoje dane w więcej niż dwóch zakresach rejestrów, więc albo nie odczytasz na raz wszystkich (może nie ma takiej potrzeby ? nie wszystkie parametry są istotne), albo można to obejść np. tworząc kilka kopii tego samego skryptu w kilu katalogach i dla każdej zdefiniować inne zakresy rejestrów w dedykowanym config.cfg. Można też trochę przerobić InverterData.py aby czytał więcej niż 2 zakresy (jeśli programujesz w pythonie, to zmiana powinna być dość prosta).
Powodzenia :)

closed 2022-03-23 by MichaluxPL

ty5ki, 2022-04-20:
Witam,
Czyli wystarczy zmienić zakresy na te z tabeli excela?
[image]

np:
requests:
  - start: 0x0000 --> 0x0480
    end:  0x0027 --> 0x04FF
    mb_functioncode: 0x03
  - start: 0x0105 --> 0x0680
    end: 0x0114 --> 0x067F
    mb_functioncode: 0x03

U mnie to nie zmieniło nic na lepsze :/
Było tak:
[image]
a po zmianie:
[image]

MichaluxPL, 2022-04-22:
Nie wystarczy.
Oprócz tego trzeba też  przypisać w SOFARMap.xml numery rejestrów do
poszczególnych wartości (w dokumentacji do MODBUSa danego falownika powinna
być informacja, która wartość jest przechowywana pod którym adresem
rejestru).

Michał