MichaluxPL/Sofar_LSW3 #37: Influxdb + grafana: Daily production panel shows negative values

perepujal, 2022-04-14:
Hi, 
the current query (in grafana.ifdb.json, about line 850) gives negative values at start of each day, it can be addressed with a trick:
` "query": "SELECT ((difference(max(\"SolarProduction_Today\")) + abs(difference(max(\"SolarProduction_Today\")))) / 2) FROM \"InverterData\" WHERE $timeFilter GROUP BY time(30m) fill(null)",`

The trick just gets the absolute value and adds it to the first value, so it zeroes the negatives, then it divides by 2 to get the original positive values again.

HTH
Pere

MichaluxPL, 2022-04-28:
Try this one:
SELECT non_negative_difference(max(\"SolarProduction_Today\")) FROM \"InverterData\" WHERE $timeFilter GROUP BY time(30m) fill(0)

perepujal, 2022-05-03:
It works, thanks :)

closed 2022-05-04 by MichaluxPL