MichaluxPL/Sofar_LSW3 #38: Sofar LSW3 - No value in response for register 0x0001

philkemp, 2022-04-27:
Hi,

Thanks for sharing your code for this :)

I'm trying to get the realtime usage statistics out of our Sofar HYD3000-ES inverter via the LSW3 wifi dongle, I've been able to get the hardware information as below:

{
    "Product code": "15kW",             <---- This happens to be incorrect :-/.  it's a 3kW Inverter :-/
    "Serial Number": "SMxxxxxxxxx",
    "Software Version": "V260",
    "Hardware Version": "V100",
    "DSP Version": "V260"
}

But when I try to extract the statistics I'm getting

No value in response for register 0x0001
Check register start/end values in config.cfg.   <--  This is left as the default register_start1=0x0000 & register_end1=0x0027 

I suspect this may be due to an issue with the Data Logger rather than the code :-/ as the Yield values are shown as 0.0 kWh despite uploading the correct figures to Solarman :-/   as per image below, think this may be a firmware issue but cant locate any firmware downloads.

[image]

Just wondering if you've seen the 'No value in response for register 0x0001' error before or have any ideas before I pursue the lack of yield information issue with Solarman / Sofar.

Many Thanks!

Phil

philkemp, 2022-04-27:
Don't know if this assists, I enabled verbose=1 in the config.cfg

Connecting to logger... connected successfully !
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x98\xc0\xbdh\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x00\x00(E\xd4\x00\x15')
Received data:  b'\xa5\x10\x00\x10\x15\x00V\x98\xc0\xbdh\x02\x01\xed\xe7&\x00\xa4\x19\x00\x00d~Bb\x05\x00M\x15'
0x0000 - Inverter status: 21
No value in response for register 0x0001
Check register start/end values in config.cfg

philkemp, 2022-04-27:
Apologies, very new to github, have located the closed issues and found the register values for HYD inverters from another post and understand how the results are mapped in the relevant XML files etc...

Unfortunately despite updating the script to use the 0x2xx range I constantly get messages like 'No value in response for register 0x0253' which should contain PV2 voltage so think the logger isn't working correctly locally despite working with Solarman.

I will close the issue and chase Sofar / Solarman for a firmware update for the LSW3 logger.

Thank you

closed 2022-04-27 by philkemp

glynnallen1704, 2023-12-18:
> Apologies, very new to github, have located the closed issues and found the register values for HYD inverters from another post and understand how the results are mapped in the relevant XML files [...]

Did you manage to get an updated firmware?