MichaluxPL/Sofar_LSW3 #39: Is it possible to add support for HYD 6000-EP inverter ?

maxxximatoze, 2022-05-02:
I tried a lot of things with no luck. No answer from the wifi logger (LSW3_15_270A_1.32)

I got things like that with too short answer:

root@nuc:~/sofarsolar/Sofar_LSW3-main# ./InverterData.py
Connecting to logger... connected successfully !
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x00\x00\x18E\xc0\x00\x15')
Received data:  b'\xa5\x13\x00\x10\x15\x00\x0f\x8d\xad,\x89\x02\x01\xc6\xa6\x00\x00~\x02\x00\x00F\x8aob\x01\x90\x02\x00\x03\\\x15'
0x0000 - Inverter status: "FAULT"
0x0001 - Fault 1: 23573
No value in response for register 0x0002
Check register start/end values in config.cfg

It would be so cool to get valid data, so I can do what I want like cool realtime graph ans stats.

MichaluxPL, 2022-05-03:
If You have manufacturer's MODBUS protocol specification for this inverter - send it to me. I can try to suggest the correct setting or required changes in the code. Unfortunately without actual logger/inverter access I won't be able to do much more :(

maxxximatoze, 2022-05-03:
Hello, thx for the answer. Where I can find this MODBUS specs for the HYD6000-EP ? May I ask to sofar solar directly ? Will they anwser to me as I'm not a professionnal ? thx a lot.

Regards.

----- Mail original -----

maxxximatoze, 2022-05-03:
If you want, I can give you access to my inverter through an openvpn account, could it interest you to look at it ?

Regards

----- Mail original -----

MichaluxPL, 2022-05-03:
>

paqp, 2022-05-06:
hello,
I have HYD6000 with lsw3.
I want to inject datas to home assistant.

Based on this PDF 
https://community.openenergymonitor.org/uploads/short-url/dF8j79PsvtCS0xQ5aPxHXO4HZDE.pdf

there are some parameters I cannot find

I edited a yaml file like this

```
>   requests:
>     - start: 0x0200
>       end:  0x021F
>       mb_functioncode: 0x03
>     - start: 0x0220
>       end: 0x0245
>       mb_functioncode: 0x03
>   
>   
>   parameters:
>    - group: solar
>      items:
>       - name: "Daily Production"
>         class: "energy"
>         state_class: "total"
>         uom: "kWh"
>         scale: 0.01
>         rule: 1
>         registers: [0x0218]
>         icon: 'mdi:solar-power'
>   
>       - name: "Total Production"
>         class: "energy"
>         state_class: "total_increasing"
>         uom: "kWh"
>         scale: 1
>         rule: 3
>         registers: [0x021D,0x021C]
>         icon: 'mdi:solar-power'
>   
>       - name: "Total generation time"
>         class: ""
>         state_class: "measurement"
>         uom: "h"
>         scale: 1
>         rule: 3
>         registers: [0x0245,0x0244]
>         icon: 'mdi:clock-outline'
>   
>       - name: "Today generation time"
>         class: ""
>         state_class: "measurement"
>         uom: "min"
>         scale: 1
>         rule: 1
>         registers: [0x0243]
>         icon: 'mdi:clock-outline'
>   
>    - group: Output
>      items:
>       - name: "Grid frequency"
>         class: "frequency"
>         state_class: "measurement"
>         uom: "Hz"
>         scale: 0.01
>         rule: 1
>         registers: [0x020C]
>         icon: 'mdi:home-lightning-bolt'
[... 366 lines cut ...]
>            value: "ID69"
>         -  key: 32
>            value: "ID70 The Output current is too high,and has cause unrecoverable fault"
>         -  key: 64
>            value: "ID71"
>         -  key: 128
>            value: "ID72 Reserved"
>         -  key: 256
>            value: "ID73 Reserved"
>         -  key: 512
>            value: "ID74"
>         -  key: 1024
>            value: "ID75 Error writing from EEPROM"
>         -  key: 2048
>            value: "ID76 Error reading to EEPROM"
>         -  key: 4096
>            value: "ID77 Relay fauilure causes unrecoverable fault"
>         -  key: 8192
>            value: "ID78 Reserved"
>         -  key: 16384
>            value: "ID79 Reserved"
>         -  key: 32768
>            value: "ID80 Reserved"
>         icon: 'mdi:wrench'
```

MichaluxPL, 2022-05-11:
We can try :)

maxxximatoze, 2022-05-11:
Here is a link to the xlsx file https://tinyurl.com/fk8yz6w7

Tell me if it is usefull, Thx !

MichaluxPL, 2022-05-12:
Unfortunately this is not the whole MODBUS specs, since no function codes has been documented :( So I don't know if Your inverter uses the same function numbers to read data as mine.
Nevertheless I'd set register ranges like that:
register_start1=0x0484
register_end1=0x04BC
register_start2=0x0504
register_end2=0x0527
registerhw_start=0x0444
registerhw_end=0x0452
Your inverter is much more complex than mine, so my script won't enable You to get all data, but let's see if You can get any valuable data at all with above config. If yes, than we can think about the next steps. If not, than I'm afraid I won't be able to help You :(

maxxximatoze, 2022-05-12:
Hello,

Here is the result with config.cfg modified accordingly to your values, SOFARMap.xml and SOFARHXMap.xml accordingly to the xlsx file.

[SofarInverter]
inverter_ip=192.168.0.230
inverter_port=8899
inverter_sn=XXXXXXXXXXXX
register_start1=0x0484
register_end1=0x04BC
register_start2=0x0504
register_end2=0x0527
registerhw_start=0x0444
registerhw_end=0x0452
lang=EN
verbose=1

***@***.***:~/sofarsolar/Sofar_LSW3-main# ./InverterData.py
Connecting to logger... connected successfully !
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00H\x009\x05\xce\x00\x15')
Received data:  b'\xa5\x13\x00\x10\x15\x00?\x8d\xad,\x89\x02\x01\xff\x13\r\x00\x06\x12\x00\x00\xbd\xe0ob\x01\x90\x02\x00\x03\xa4\x15'
0x0484 - Grid frequency: 0.03Hz
0x0485 - Output active power: -235310W
No value in response for register 0x0486
Check register start/end values in config.cfg

and

***@***.***:~/sofarsolar/Sofar_LSW3-main# ./InverterHWData.py
Modbus request: 0104 0444 000f f12b
Hex string to send: A5 1700 1045 0000 8dad2c89 020000000000000000000000000000 01040444000fbf02 00 15
Data sent:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x04\x04D\x00\x0f\xf1+\x00\x15')
Data received:  b'\xa5\x10\x00\x10\x15\x00@\x8d\xad,\x89\x02\x01R\x14\r\x00Y\x12\x00\x00\xbd\xe0ob\x05\x00\xb8\x15'
Hex string received: A5 10 00 10 15 00 40 8D AD 2C 89 02 01 52 14 0D 00 59 12 00 00 BD E0 6F 62 05 00 B8 15
Traceback (most recent call last):
  File "/root/sofarsolar/Sofar_LSW3-main/./InverterHWData.py", line 121, in <module>
    val2=chr(int(str(responsereg[2:4]),16))
ValueError: invalid literal for int() with base 16: ''

The inverter answer seems too short... No luck ! If you have another trick to test, tell me, I will do the test with pleasure. Thx.

----- Mail original -----

MichaluxPL, 2022-05-13:
Yep, You haven't got valid data from the logger/inverter yet.
I have a few more ideas, but I'll need direct access to test those things.
If You can arrange it, I can try a few more things.

MichaluxPL, 2022-05-13:
Give me Your e-mail, and I'll contact You directly.

maxxximatoze, 2022-05-14:
maxime_pierron79/free_fr , replace underscores by dots and slash by at...

----- Mail original -----

maxxximatoze, 2022-05-14:
Here it is, a wireguard config file to import in your wireguard client, if you don't have it go https://www.wireguard.com/install/ to install it. If you prefer openvpn tell me I also have a server.

The inverter's ip is 192.168.0.230

If you need a local linux server to make your access/tests localy, I can provide you access to a debian or a ubuntu server vm to do whatever you want. Just tell me. Thx.

Regards.

----- Mail original -----

maxxximatoze, 2022-05-14:
Last but not least, the wireguard server will be up on your demand, so when you want to connect, tell me and I up the server asap. thx !

Regards.

----- Mail original -----

MichaluxPL, 2022-05-14:
First of all, there is no attachment :(
Second - I'd strongly prefer openvpn (we've been talking about this one
before - not wireguard).

Michal

maxxximatoze, 2022-05-14:
OK, that's weird because in my email the attechement is here. No problem. Here is the openvpn config file (PW MichaluxPL). Tell me if there is a problem what so ever. Thx.

Regards.

----- Mail original -----

Michał

isluxor, 2022-06-05:
What is the configuration for enable the HYD 6000 inverter?

jaimelaborda, 2022-07-05:
I'm also interested!

MichaluxPL, 2022-07-05:
I've tried a few tricks together with @maxxximatoze to adapt my solution for HYD inverters, but we didn't have any luck.
Unfortunately I don't own such an inverter and have so much time to drill it down ...
So currently no support for this inverter type, sorry about that.

closed 2022-07-05 by MichaluxPL

maxxximatoze, 2022-10-11:
> Due to the differences between inverter models, some parameters can be indeed missing. You can replace them with the different ones Your inverter has and my does not.
> According to attached modbus specs, I'd set register ranges like this:
> registerhw_start=0x2000
> registerhw_end=0x200B
> register_start1=0x0200
> register_end1=0x0220
> register_start2=0x0221
> register_end2=0x0245
> Of course SOFARMap.xml has to be changed accordingly to reflect above register and their meanig (according to the modbus doc).
> One more thing - in the original script there are two parameters hard coded, that use two registers (2 bytes) for each to keep the whole value.
> For my inverter those are: Total production (0x0015-0x0016) and Total generation time (0x0017-0x0018).
> If read it correctly, for Your inverter, those values are being kept in registers 0x021C-0x021D and 0x0244-0x0245 respectively.
> Also, as I can see, Your inverter has more parameters that 2-byte values. In order to get the correct final values it should be reflected in the script (in the similar way like for those 2 above).
> Anyway I'd start with setting above register ranges to see if You can get the correct answer length, which would indicate, You get actual values in response. Until then, any other changes can wait :)

Here is what sofar solar sent to me. It's in chinese but it should be usable. I need time to look at it and do some tests.

----- Mail original -----

perepujal, 2022-11-06:
Hi, I start to get some valid data, but I still have some problems,

First the modbus protocol I use for HYD6000EP is the one at 
https://github.com/pixellos/codereinvented.automation.py/blob/master/SOFAR%20HYD-3PH%20and%20SOFAR%20-G3%20Modbus%20Protocol%202021-10-14_Client.xlsx

I have modified the xml file to match some of the fields, not all currently, just for testing.

Looking at the protocol and setting the following values do not give any valuable result
register_start1=0x0480
register_end1=0x04FF
register_start2=0x0580
register_end2=0x05FF

Instead, if I manually set the business field inside the program, then I start to get correct values:
    businessfield= binascii.unhexlify('01030480003F') # ON GRID: Grid frequency, output active power...
or
    businessfield= binascii.unhexlify('01030580003F') #  PV INPUT...  Currently I get 0W for 0x0586 PW1 Power but I do get PV1 voltage and current
Note the 0480 and 0580 start addresses.

I imagine somehow pos_ini and/or pos_end do not get the correct values and the inverter refuses the request.

There will be also need for more register starts/ends, for battery status, offgrid,... but I think this could wait for now

HTH, and thanks for any comments

Pere

perepujal, 2022-11-07:
Some progress, registers should be 
register_end1=0x04BE
register_end2=0x05BF
the xml file is still unfinished
In plus, I had to tweak a little how pos_ini, pos_fin, businessfield and crc are generated, hope this doesn't break anything, my python is very basic...
[InverterData.py.diff.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/9947132/InverterData.py.diff.txt)
Whith this patch I get  2 ranges of registers(PV and ongrid), now I need to find a way to get the other needed ranges...

HTH
Pere

perepujal, 2022-11-17:
Hi, I am attaching what half works for me, the connection with the inverter is bad, I run a script to get the data every minute but there are many times that 20 minutes or more pass without a valid reply. In plus, when there are not battery charge nor solar power, then the inverter is nearly unconnectable, and when it replies, some values are far from correct.

[config-HYD6000EP.cfg.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10026408/config-HYD6000EP.cfg.txt)
[InverterData.py.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10026409/InverterData.py.txt)
[SOFARMap-HYD6000EP.xml.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10026410/SOFARMap-HYD6000EP.xml.txt)

Rename the attached files to config.cfg, SOFARMap.xml and InverterData.py, I can't attach them with their original extensions, edit config.cfg to match your system and then run InverterData.py when there is solar power or when the batteries have some charge, otherwise you will not get a correct reply from the inverter.
Most likely it will fail in the first time, and the second and third, but eventually it will start responding, if it gets stuck on a chunk, hit CTRL C and start it again.

HTH
Pere

maxxximatoze, 2022-11-17:
> Hi, I am attaching what half works for me, the connection with the inverter is bad, I run a script to get the data every minute but there are many times that 20 minutes or more pass without a valid [...]

Hi,

Just tried your modified config file and script and that's not bad at all, it really could be usable with a little polishing ! here is the result on first try:

Connecting to logger... connected successfully !
chunks  0
1024 1082 0x3a 0x3a posini  0x0400   posfin  0x003b
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x00\x00;\x05)\x00\x15')
Received data:  b'\xa5\x89\x00\x10\x15\x00T\x8d\xad,\x89\x02\x01a\xa9\x848\x87\x0c\x00\x000j\xf1*\x01\x03v\x00\x03\xf9\xc1\x05\x81\xff\xff\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00%\x00\x00\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xe1\x00\x03\t\x9e\x00\x041q\x01\xaa\x00\x16\x00\x0b\x00\x11\x00\x0b\x009\x00\x14\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00"\xc4E\x15'
0x0400 - SysInfo datacheck: 3err
0x0404 - Inverter status: "Normal"
0x0405 - Fault 1: "No error"
0x0406 - Fault 2: "No error"
0x0407 - Fault 3: "No error"
0x0408 - Fault 4: "No error"
0x0409 - Fault 5: "No error"
0x0421 - Inverter module temperature: 0ºC
0x0422 - Inverter inner termperature: 0ºC
0x0426 - Today generation time: 225min
0x0427 - Total generation time: 3m
0x0428 - Total generation time: 2462m
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAa
chunks  1
1152 1214 0x3e 0x3e posini  0x0480   posfin  0x003f
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x80\x00?\x05\x02\x00\x15')
Received data:  b'\xa5\x91\x00\x10\x15\x00U\x8d\xad,\x89\x02\x01b\xa9\x848\x88\x0c\x00\x000j\xf1*\x01\x03~\x00\x00\xc0\x00\x00\x04a?\x13\x88\x00?\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\t`\x01\x10\x00\x00\x00\x00\x00\x00\x00i\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00?\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00e\xf5I\x15'
0x0480 - Output datacheck: 0errors
0x0484 - Grid frequency: 50.0Hz
0x0485 - Output active power: 630W
0x0486 - Output reactive power: 0kVar
0x0488 - Grid frequency: 0.0W
0x04B0 - L1 Voltage: 0.0V
0x04B0 - L1 Voltage: 0.0V
0x04B1 - L1 Current: 0.0A
0x04B1 - L1 Current: 0.0A
0x04B5 - L2 Voltage: 0.0V
0x04B5 - L2 Voltage: 0.0V
0x04B6 - L2 Current: 0.0A
0x04B6 - L2 Current: 0.0A
BBBBBBBBBBBBBBBBBBBBB
chunks  2
1280 1343 0x3f 0x3f posini  0x0500   posfin  0x0040
*** Chunk no:  2
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x05\x00\x00@D\xf6\x00\x15')
Received data:  b'\xa5\x93\x00\x10\x15\x00V\x8d\xad,\x89\x02\x01b\xa9\x848\x88\x0c\x00\x000j\xf1*\x01\x03\x80\x00\x00\x00\x00\x00\x00\x1c\x8f\x00\x00\x00\x00\x00\x00\x13\x88\x00\x00\x00\x00\t]\x00\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb3\r\x0f\x15'
0x0500 - Output datacheck: 0errors
0x0504 - Offgrid Output active power: 0W
0x0505 - OffgridOutput reactive power: 0kVar
0x0507 - OffGrid frequency: 50.0Hz
chunks  3
1408 1471 0x3f 0x3f posini  0x0580   posfin  0x0040
*** Chunk no:  3
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x05\x80\x00@E\x1e\x00\x15')
Received data:  b'\xa5\x93\x00\x10\x15\x00W\x8d\xad,\x89\x02\x01c\xa9\x848\x88\x0c\x00\x000j\xf1*\x01\x03\x80\x00\x00\x00\x00\x00\x00\x03\xff\x08\xa8\x00S\x00\x12\x0b\x96\x00\x97\x00,\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00>\xe21\x15'
0x0580 - PV check: 0W
0x0584 - PV1 Voltage: 221.6V
0x0585 - PV1 Current: 0.83A
0x0586 - PV1 Power: 180W
0x0587 - PV2 Voltage: 296.6V
0x0588 - PV2 Current: 1.51A
0x0589 - PV2 Power: 440W
chunks  4
1472 1523 0x33 0x33 posini  0x05c0   posfin  0x0034
*** Chunk no:  4
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x05\xc0\x004D\xed\x00\x15')
Received data:  b'\xa5{\x00\x10\x15\x00X\x8d\xad,\x89\x02\x01c\xa9\x848\x89\x0c\x00\x000j\xf1*\x01\x03h\x00\x00\x00\x00\x00\x00\x00\x0f\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0 [... cut ...]
  File "/usr/lib/python3/dist-packages/requests/sessions.py", line 655, in send
    r = adapter.send(request, **kwargs)
  File "/usr/lib/python3/dist-packages/requests/adapters.py", line 516, in send
    raise ConnectionError(e, request=request)
requests.exceptions.ConnectionError: HTTPConnectionPool(host='192.168.1.33', port=8086): Max retries exceeded with url: /write?db=SofarDB (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fccbf2acb50>: Failed to establish a new connection: [Errno 113] No route to host'))

maxxximatoze, 2022-11-17:
I have oversimplified the invertdata script, remove verbose and here is the result:

root@nuc:~/sofarsolar/Sofar_LSW3-main/test# time ./InverterData.py
chunks  0
1024 1082 0x3a 0x3a posini  0x0400   posfin  0x003b
chunks  1
1152 1214 0x3e 0x3e posini  0x0480   posfin  0x003f
chunks  2
1280 1343 0x3f 0x3f posini  0x0500   posfin  0x0040
chunks  3
1408 1471 0x3f 0x3f posini  0x0580   posfin  0x0040
chunks  4
1472 1523 0x33 0x33 posini  0x05c0   posfin  0x0034
chunks  5
1536 1595 0x3b 0x3b posini  0x0600   posfin  0x003c
chunks  6
1600 1631 0x1f 0x1f posini  0x0640   posfin  0x0020
chunks  7
1664 1690 0x1a 0x1a posini  0x0680   posfin  0x001b
*** JSON output:
{
    "SysInfo datacheck (err)": 3,
    "Inverter status": "Normal",
    "Fault 1": "No error",
    "Fault 2": "No error",
    "Fault 3": "No error",
    "Fault 4": "No error",
    "Fault 5": "No error",
    "Inverter module temperature (ºC)": 0,
    "Inverter inner termperature (ºC)": 0,
    "Today generation time (min)": 260,
    "Total generation time (m)": 2497,
    "Output datacheck (errors)": 0,
    "Grid frequency (Hz)": 50.01,
    "Output active power (W)": 620,
    "Output reactive power (kVar)": 0,
    "Grid frequency (W)": -0.01,
    "L1 Voltage (V)": 0.0,
    "L1 Current (A)": 0.0,
    "L2 Voltage (V)": 0.0,
    "L2 Current (A)": 0.0,
    "Offgrid Output active power (W)": 0,
    "OffgridOutput reactive power (kVar)": 0,
    "OffGrid frequency (Hz)": 50.0,
    "PV check (W)": 0,
    "PV1 Voltage (V)": 226.2,
    "PV1 Current (A)": 0.35,
    "PV1 Power (W)": 50,
    "PV2 Voltage (V)": 292.8,
    "PV2 Current (A)": 0.91,
    "PV2 Power (W)": 200,
    "Voltage Bat1 (V)": 5.16,
    "Current Bat1 (A)": -5.43,
    "Power Bat1 (W)": -0.28,
    "soc (percent)": 0.84,
    "Cycles (cycles)": 0,
    "Stats datacheck (err)": 0,
    "Today production (Wh)": 2210,
    "Total production (kWh)": 2883.5
}

real    0m2.596s
user    0m1.012s
sys     0m0.067s

after several runs, no error, it connects, gets the data, and display them. Very nice. The output seems to have some bad naming like "Grid frequency (W)": -0.01, should be grid power or  "PV check (W)": 0, don't know what is it or  "Power Bat1 (W)": -0.28, which should be kW but nothing more than that.

maxxximatoze, 2022-11-17:
Question, my inverter always returns temps at 0°C, is it the same for you ?

    "Inverter module temperature (ºC)": 0,
    "Inverter inner termperature (ºC)": 0,

maxxximatoze, 2022-11-17:
After some minors adjustements, here is what I get, all reported values seem to be good execpt for "Total generation time (m)": 2579, not sure for this one

root@nuc:~/sofarsolar/Sofar_LSW3-main/test# time ./InverterData.py
1024 1082 0x3a 0x3a posini  0x0400   posfin  0x003b
1152 1214 0x3e 0x3e posini  0x0480   posfin  0x003f
1280 1343 0x3f 0x3f posini  0x0500   posfin  0x0040
1408 1471 0x3f 0x3f posini  0x0580   posfin  0x0040
1472 1523 0x33 0x33 posini  0x05c0   posfin  0x0034
1536 1595 0x3b 0x3b posini  0x0600   posfin  0x003c
1600 1631 0x1f 0x1f posini  0x0640   posfin  0x0020
1664 1690 0x1a 0x1a posini  0x0680   posfin  0x001b
*** JSON output:
{
    "SysInfo datacheck (err)": 3,
    "Inverter status": "Normal",
    "Fault 1": "No error",
    "Fault 2": "No error",
    "Fault 3": "No error",
    "Fault 4": "No error",
    "Fault 5": "No error",
    "Inverter module temperature (ºC)": 0,
    "Inverter inner termperature (ºC)": 0,
    "Today generation time (min)": 342,
    "Total generation time (m)": 2579,
    "Output datacheck (errors)": 0,
    "Grid Frequency (Hz)": 49.95,
    "Output active power (W)": 650,
    "Output reactive power (kVar)": 0,
    "Grid Power (W)": -30,
    "L1 Voltage (V)": 0.0,
    "L1 Current (A)": 0.0,
    "L2 Voltage (V)": 0.0,
    "L2 Current (A)": 0.0,
    "Offgrid Output active power (W)": 0,
    "Offgrid Output reactive power (kVar)": 0,
    "OffGrid frequency (Hz)": 49.96,
    "PV check (W)": 0,
    "PV1 Voltage (V)": 241.0,
    "PV1 Current (A)": 1.27,
    "PV1 Power (W)": 300,
    "PV2 Voltage (V)": 296.5,
    "PV2 Current (A)": 2.03,
    "PV2 Power (W)": 600,
    "Voltage Bat1 (V)": 52.3,
    "Current Bat1 (A)": 2.56,
    "Power Bat1 (W)": 130,
    "soc (percent)": 0.85,
    "Cycles (cycles)": 0,
    "Stats datacheck (err)": 0,
    "Today production (kWh)": 3.34,
    "Total production (kWh)": 2884.6
}

real    0m2.518s
user    0m0.979s
sys     0m0.099s

maxxximatoze, 2022-11-17:
I have added some fields (see below) and corrected temps, but load consumption total and bat discharge total show bad values...

root@nuc:~/sofarsolar# time ./InverterData.py
1024 1082 0x3a 0x3a posini  0x0400   posfin  0x003b
1152 1214 0x3e 0x3e posini  0x0480   posfin  0x003f
1280 1343 0x3f 0x3f posini  0x0500   posfin  0x0040
1408 1471 0x3f 0x3f posini  0x0580   posfin  0x0040
1472 1523 0x33 0x33 posini  0x05c0   posfin  0x0034
1536 1595 0x3b 0x3b posini  0x0600   posfin  0x003c
1600 1631 0x1f 0x1f posini  0x0640   posfin  0x0020
1664 1690 0x1a 0x1a posini  0x0680   posfin  0x001b
*** JSON output:
{
    "SysInfo datacheck (err)": 3,
    "Inverter status": "Normal",
    "Fault 1": "No error",
    "Fault 2": "No error",
    "Fault 3": "No error",
    "Fault 4": "No error",
    "Fault 5": "No error",
    "Inverter ambient temperature (ºC)": 36,
    "Inverter radiator termperature (ºC)": 21,
    "Today generation time (min)": 614,
    "Total generation time (min)": 2851,
    "Output datacheck (errors)": 0,
    "Grid Frequency (Hz)": 50.0,
    "Output active power (W)": 420,
    "Output reactive power (kVar)": 0,
    "Grid Power (W)": 0,
    "L1 Voltage (V)": 0.0,
    "L1 Current (A)": 0.0,
    "L2 Voltage (V)": 0.0,
    "L2 Current (A)": 0.0,
    "Offgrid Output active power (W)": 0,
    "Offgrid Output reactive power (kVar)": 0,
    "Offgrid Frequency (Hz)": 50.0,
    "PV check (W)": 0,
    "PV1 Voltage (V)": 20.1,
    "PV1 Current (A)": 0.01,
    "PV1 Power (W)": 0,
    "PV2 Voltage (V)": 27.9,
    "PV2 Current (A)": 0.0,
    "PV2 Power (W)": 0,
    "PV Total Power (W)": 0,
    "Bat1 Voltage (V)": 51.3,
    "Bat1 Current (A)": -8.84,
    "Bat1 Power (W)": -450,
    "SOC (percent)": 0.85,
    "Cycles (cycles)": 0,
    "Stats datacheck (err)": 0,
    "PV Generation Today (kWh)": 4.78,
    "PV Generation Total (kWh)": 2886.1,
    "Load Consumption Today (kWh)": 9.63,
    "Load Consumption Total (kWh)": -3206.0,
    "Energy Purchase Today (kWh)": 5.82,
    "Energy Purchase Total (kWh)": 624.1,
    "Energy Selling Today (kWh)": 0.0,
    "Energy Selling Total (kWh)": 3.3,
    "Bat Charge Today (kWh)": 1.75,
    "Bat Charge Total (kWh)": 689.0,
    "Bat Discharge Today (kWh)": 0.98,
    "Bat Discharge Total (kWh)": 0.0
}

real    0m2.635s
user    0m1.073s
sys     0m0.093s

perepujal, 2022-11-18:
Hi, glad you find it useful :)

Did you attach the modified files? I didn't see them, if you attached them to a email, then most likely github removed them before storing it in the issue's web page or resending to people following the issue. Github, at least in the web interface don't like some extensions in attachments :/

About load consumption and bat discharge totals, I imagine their register line should be something like
"registers": ["0x068A","0x068B"],
"registers": ["0x069A","0x069B"],
If that don't fix the values, then no idea :(  the first field of each range should give a hint about the correctness of the range values... see the xlsx file

maxxximatoze, 2022-11-18:
Hi,

for 68a/b and 69a/b it is what I've done, but for these to values, the output is not good but at least there is a value. I don't know why this happening. Need more tests.

here is the modified files of my test
[InverterData.py.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10040100/InverterData.py.txt)
[config.cfg.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10040103/config.cfg.txt)
[SOFARMap.xml.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10040104/SOFARMap.xml.txt)

maxxximatoze, 2022-11-19:
As you can see on this png graph, I also have the problem two times a day with the inverter not responding to query with no reason, weird
[image]

perepujal, 2022-11-19:
Just two times a day, you are lucky ;)  I either have a faulty dongle or a faulty local network but the lacks of data are much more than yours.

As it is currently, the script makes eight queries in one connection, if any of them fails, the script fails.
I am considering to split the queries in 8 connections, only one query each connection, that way even if one query fails, the others could possibly be fine...

maxxximatoze, 2022-11-19:
Not sure that it is a good idea because as it is, the success rate is almost perfect and it is fast, 2.5 seconds to retreive data. The problem seems to be different because, as it can be seen on the graph, there two period of time, in the morning and in the evening, where the inverter does not respond at all, this behavior let me think about a firmware bug and/or limitation. I really would like to test with an ethernet adapter instead of wifi, but the usb version compatible with the hyd600-ep is not easy to find, the only one available is the serial one, too bad...

maxxximatoze, 2022-11-21:
Here is my little real time online status for my inverter, works pretty well

SofarSolar real time status on 21-11-2022 18:45:22 (refreshed every 10 seconds)

Inverter status: Normal
Fault 1: No error
Fault 2: No error
Fault 3: No error
Fault 4: No error
Fault 5: No error

Inverter ambient temperature (ÂºC): 35
Inverter radiator termperature (ÂºC): 20

Grid Frequency (Hz): 49.99
Offgrid Frequency (Hz): 49.99

Load Power (W): 310
PV Power (W): 0
Bat1 Power (W): -320
Grid Power (W): -30
Output active power (W): 280
Offgrid Output active power (W): 0

PV1 Voltage (V): 50.0
PV1 Current (A): 0.03
PV1 Power (W): 0
PV2 Voltage (V): 62.1
PV2 Current (A): 0.0
PV2 Power (W): 0

Bat1 Voltage (V): 51.0
Bat1 Current (A): -6.39
SOC (percent): 0.46

PV Generation Today (kWh): 3.13
Load Consumption Today (kWh): 10.42
Energy Selling Today (kWh): 0.0
Energy Purchase Today (kWh): 5.35
Bat Charge Today (kWh): 0.77
Bat Discharge Today (kWh): 3.12

One thing is that when the inverter is offgrid (for whatever reason) it becomes unaccessible, and so no more online status or live graph update. I really need to test with the ethernet adapter.

maxxximatoze, 2022-11-25:
Here is the last version with some modifications. works pretty well :)

For information, if Inverter datacheck (error)":  is different of 3 then the values are bad so you can can skip them. Also, when the inverter has no power source (solar or battery) it gets offline and do not respond to network (the solarman smart website is not updated either), so there is not much that we can do about it.

[SOFARMap.xml.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10093506/SOFARMap.xml.txt)
[config.cfg.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10093507/config.cfg.txt)
[InverterData.py.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10093508/InverterData.py.txt)

maxxximatoze, 2022-11-25:
Interesting thing, when the inverter is deconnected from the grid, InverterData.py is entering a endless loop, sending the frame to get the register 0X0480 which is corresponding to "on grid output" without getting an anwser... Here is the problem.

root@nuc:~/sofarsolar# time ./InverterData.py
Connecting to logger... connected successfully !
1024 1087 0x3f 0x3f posini  0x0400   posfin  0x0040
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x00\x00@E\n\x00\x15')
Received data:  b'\xa5\x93\x00\x10\x15\x00\x9d\x8d\xad,\x89\x02\x01\x19n\x8f8\xaf\x02\x00\x00pl\xf1*\x01\x03\x80\x00\x00\xc0\x00\x00\x04a?\x00\x00\xff\xfe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00,\x00\x00\x00\x00\x00\x00\x00{\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xff\xfe\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xda\x16\xb7\x15'
0x0400 - Inverter datacheck: 0error
0x0404 - Inverter status: "Stand-by"
0x0418 - Inverter ambient temperature: 0ºC
0x041A - Inverter radiator termperature: 0ºC
1152 1215 0x3f 0x3f posini  0x0480   posfin  0x0040
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x80\x00@D\xe2\x00\x15')
1152 1215 0x3f 0x3f posini  0x0480   posfin  0x0040
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x80\x00@D\xe2\x00\x15')
1152 1215 0x3f 0x3f posini  0x0480   posfin  0x0040
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x80\x00@D\xe2\x00\x15')
1152 1215 0x3f 0x3f posini  0x0480   posfin  0x0040
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x80\x00@D\xe2\x00\x15')
1152 1215 0x3f 0x3f posini  0x0480   posfin  0x0040
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x80\x00@D\xe2\x00\x15')
1152 1215 0x3f 0x3f posini  0x0480   posfin  0x0040
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x80\x00@D\xe2\x00\x15')
1152 1215 0x3f 0x3f posini  0x0480   posfin  0x0040
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x80\x00@D\xe2\x00\x15')
1152 1215 0x3f 0x3f posini  0x0480   posfin  0x0040
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x80\x00@D\xe2\x00\x15')
1152 1215 0x3f 0x3f posini  0x0480   posfin  0x0040
*** Chunk no:  1
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x80\x00@D\xe2\x00\x15')
1152 1215 0x3f 0x3f posini  0x0480   posfin  0x0040
*** Chunk no:  1

maxxximatoze, 2022-11-25:
Here a modified version that seems to work when inverted is offgrid...

[InverterData.py.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10093996/InverterData.py.txt)
[SOFARMap.xml.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10093997/SOFARMap.xml.txt)
[config.cfg.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10093998/config.cfg.txt)

perepujal, 2023-01-05:
[maxxximatoze](https://github.com/maxxximatoze) wrote
>  Also, when the inverter has no power source (solar or battery) it gets offline and do not respond to network (the solarman smart website is not updated either), so there is not much that we can do about it.

I got some success, just tested a couple of days, so nothing definitive
The main thing I want to know when the inverter is out of power is the grid import, so I created a minimalistic config file that just points to 
register_start1=0X0480
register_end1=0X04BC
and luckily it worked fine over two nights, of course I had too to modify the .py script so it allows just one register range and that it allows to be called with the name of the config file like
./InverterData.py configminimal.cfg
if you just call ./InverterData.py without parameters it will use config.cfg as usual
[InverterDataImportcfg.py.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10356359/InverterDataImportcfg.py.txt)

maxxximatoze, 2023-01-06:
> [maxxximatoze](https://github.com/maxxximatoze) wrote
>
>
> I got some success, just tested a couple of days, so nothing definitive The main thing I want to know when the inverter is out of power is the [...]

Nice.
I also got some success with this version I'm using since december. When the inverter status change to standby because of empty battery no solar (night), I just reduce the polling interval to every 30s instead of every 10s and it works just fine, at least for me. Almost no error getting data from the inverter.

You can see on this graph below, from 00h00 to about 8H30 when the sun comes up, the grid power is here. For what it is meant to be use, it is just what I need for a long time. Thanks again for providing your script.

[image]

[InverterData.py.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10360530/InverterData.py.txt)
[SOFARMap.xml.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10360531/SOFARMap.xml.txt)
[config.cfg.txt](https://github.com/MichaluxPL/Sofar_LSW3/files/10360532/config.cfg.txt)

maxxximatoze, 2023-04-07:
> [maxxximatoze](https://github.com/maxxximatoze) wrote
>
>
> I got some success, just tested a couple of days, so nothing definitive The main thing I want to know when the inverter is out of power is the [...]

Hello perepujal,

I'm using your python script for months now and it's working really great throught the LSW3 wifi adapter. I have just bought a LSE3 usb ethernet adapter and, no luck, the Inverter.py script does not work with it. I have edited the serial num in config.cfg, the LSE3 is listening on port 8899 as LSW3, but the script receive no data, bellow the output:

root@nuc:~/sofarsolar# ./InverterData.py
1024 1087 0x3f 0x3f posini  0x0400   posfin  0x0040
No value in response for register 0x0400
Check register start/end values in config.cfg

The LSE3 adatper was in firmware 1.08, so I've tried to upgrade it to 1.09 with the same result. It does not seem to work the same as LSW3...

Do you have an idea of what I can modify to try to make it works ? Edit the register values ? Thanks in advance !

Best regards.

perepujal, 2023-04-10:
No idea, if after having checked the basic, IP, port, network connectivity you find nothing, then I would enable verbose = 1 in the config file to see if I get some hints from it.

maxxximatoze, 2023-04-11:
Hello,

Thx for the answer. I have tried to enable the verbose mode and here is the result:

***@***.***:~/sofarsolar# time ./InverterData.py
Connecting to logger... connected successfully !
1024 1087 0x3f 0x3f posini  0x0400   posfin  0x0040
*** Chunk no:  0
Sent data:  ***@***.***\n\x00\x15')
Received data:  b'\xa5\x17\x00\x10E\x03\x00\xc3\x02'
No value in response for register 0x0400
Check register start/end values in config.cfg

real    0m1.189s
user    0m0.114s
sys     0m0.018s

And just for reference, here is the result with the wifi logger:

***@***.***:~/sofarsolar# time ./InverterData.py
Connecting to logger... connected successfully !
1024 1087 0x3f 0x3f posini  0x0400   posfin  0x0040
*** Chunk no:  0
Sent data:  ***@***.***\n\x00\x15')
Received data:  b'\xa5\x93\x00\x10\x15\x00\x11\x8d\xad,\x89\x02\x01N\xdd\x808a\x1d\x00\x00V2\xb4+\x01\x03\x80\x00\x03\xff\xc0\x05\x81\xff\xff\x00\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02\x00(\x00\x00\x00\x18\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x02G\x00\x05\x1a\xe1\x00\x073C\x01\xd0\x00\x17\x00\x04\x00\x0b\x00\t\x00+\x00\x1e\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00w\xd1\xe8\x15'
0x0400 - Inverter datacheck: 3error
0x0404 - Inverter status: "Normal"
0x0418 - Inverter ambient temperature: 40ºC
0x041A - Inverter radiator termperature: 24ºC
1152 1215 0x3f 0x3f posini  0x0480   posfin  0x0040
*** Chunk no:  1
Sent data:  ***@***.***\xe2\x00\x15')
Received data:  b'\xa5\x93\x00\x10\x15\x00\x12\x8d\xad,\x89\x02\x01O\xdd\x808b\x1d\x00\x00V2\xb4+\x01\x03\x80\x00\x00\xc0\x00\x00\x04a?\x13\x86\x01\x00\x00\x00\x00\x00\xff\xfb\x00\x00\x00\x00\x00\x00\x00\x00\tq\x04(\x00\x00\x00\x00\x00\x00\x00l\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x05\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00F\xd66\x15'
0x0480 - Output datacheck: 0error
0x0484 - Grid Frequency: 49.98Hz
0x0485 - Output active power: 2560W
0x0488 - Grid Power: -50W
0x048D - Phase R Voltage: 241.7V
0x048E - Phase R Current: 10.64A
1280 1343 0x3f 0x3f posini  0x0500   posfin  0x0040
*** Chunk no:  2
Sent data:  ***@***.***\xf6\x00\x15')
Received data:  b'\xa5\x93\x00\x10\x15\x00\x13\x8d\xad,\x89\x02\x01O\xdd\x808b\x1d\x00\x00V2\xb4+\x01\x03\x80\x00\x00\x00\x00\x00\x00\x0c\x8f\x00\x00\x00\x00\x00\x00\x13\x86\x00\x00\x00\x00\to\x00\x15\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00T\xd9\xf9\x15'
0x0500 - Offgrid Output datacheck: 0error
0x0504 - Offgrid Output active power: 0W
0x0507 - Offgrid Frequency: 49.98Hz
1408 1471 0x3f 0x3f posini  0x0580   posfin  0x0040
*** Chunk no:  3
Sent data:  ***@***.***\x1e\x00\x15')
Received data:  b'\xa5\x93\x00\x10\x15\x00\x14\x8d\xad,\x89\x02\x01O\xdd\x808b\x1d\x00\x00V2\xb4+\x01\x03\x80\x00\x00\x00\x00\x00\x00\x03\xff\tG\x00`\x00\x16\x0e\xb3\x00v\x00,\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00+\x02d\x15'
0x0580 - Solar datacheck: 0error
0x0584 - PV1 Voltage: 237.5V
0x0585 - PV1 Current: 0.96A
0x0586 - PV1 Power: 220W
0x0587 - PV2 Voltage: 376.3V
0x0588 - PV2 Current: 1.18A
0x0589 - PV2 Power: 440W
1472 1535 0x3f 0x3f posini  0x05c0   posfin  0x0040
*** Chunk no:  4
Sent data:  ***@***.***\xca\x00\x15')
Received data:  b'\xa5\x93\x00\x10\x15\x00\x15\x8d\xad,\x89\x02\x01P\xdd\x808c\x1d\x00\x00V2\xb4+\x01\x03\x80\x00\x00\x00\x00\x00\x00\x00\x1f\x00\x06\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\xb1\x9b\x80\x15'
0x05C4 - PV Total Power: 60W
1536 1599 0x3f 0x3f posini  0x0600   posfin  0x0040
*** Chunk no:  5
Sent data:  ***@***.***\xb2\x00\x15')
Received data:  b'\xa5\x93\x00\x10\x15\x00\x16\x8d\xad,\x89\x02\x01P\xdd\x808c\x1d\x00\x00V2\xb4+\x01\x03\x80\x00\x00\x00\x00\x00\x00\x05\xff\x02\t\xf1#\xff:\x00\x00\x000\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00 [... cut ...]
    "Battery Discharge Total (kWh)": 1083.6,
    "Internal Infos datacheck (error)": 0,
    "Leakage Current (mA)": 7,
    "Total BUS Voltage  (V)": 374.1,
    "LLC Bus Voltage (V)": 330.3,
    "Buck Boost Current (A)": -5.34
}

real    0m2.781s
user    0m0.790s
sys     0m0.048s

Don't kown what is wrong with the ethernet adapter. Maybe the first sequence of byte is wrong ?

LSE3: Sent data:  ***@***.***\n\x00\x15')
LSW3: Sent data:  ***@***.***\n\x00\x15')

Best regards.

----- Mail original -----

maxxximatoze, 2023-04-11:
Github is auto editing text before and after @ so here the first sequence with @ replaced by !

LSE3: Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\xc3Wo}\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x00\x00!E\n\x00\x15')
LSW3: Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\x8d\xad,\x89\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x04\x00\x00!E\n\x00\x15')

Best regards.

----- Mail original -----

maxxximatoze, 2023-04-17:
> Github is auto editing text before and after @ so here the first sequence with @ replaced by ! LSE3: Sent data: [...]

Hello, problem solved, it is the firmware version, I have flashed my LSE3 USB with ME_08_2701_2.06Ubin and it is working... Thx !