MichaluxPL/Sofar_LSW3 #40: Which python version is required?

pixellos, 2022-05-07:
Hi,

on Python 3.10.4 it seems to return strange errors

[image]

MichaluxPL, 2022-05-08:
Hi,

You haven;t received a correct response from the logger/inverter (it's way too short - Data received need to contain a lot more data). Script throws an error because it can't convert a character, that won't normally appear in a correct response.
It's most probably an indication of wrong register rages configuration.

pixellos, 2022-05-08:
Would it be safe to do scanner, that will try all addresses, or there is possibility that some addresses will try to write/change sth?

pixellos, 2022-05-11:
Hi, got specs, 

As 
[image]
I set up registerhw_start, end to
```
registerhw_start=0x0400
registerhw_end=0x047F
```
> SOFARHWMap.xml
```
[
  {
    "directory": "solar",
    "items": [
      {
        "titleEN": "Software Version",
        "titlePL": "Wersja oprogramowania",
        "registers": ["0x0453","0x0454"],
        "optionRanges": [],
        "value_type": "SV"
      },
      {
        "titleEN": "Hardware Version",
        "titlePL": "Wersja sprzętu",
        "registers": ["0x44D","0x44E"],
        "optionRanges": [],
        "value_type": "HV"
      }
    ]
  }
]
```

[SofarHYD.ES.ME3000SP.Modbus.protocol.pdf](https://github.com/MichaluxPL/Sofar_LSW3/files/8669788/SofarHYD.ES.ME3000SP.Modbus.protocol.pdf)
Those are ranges i used. 
[image]

But still 
```
Data sent:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\xd8<Xh\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x04\x04@\x00@\xf1\x1e\x00\x15')
Data received:  b'\xa5\x10\x00\x10\x15\x00g\xd8<Xh\x02\x01\xce\x8a@\x011\x19\x00\x00\xf0";a\x05\x00\t\x15'
Hex string received: A5 10 00 10 15 00 67 D8 3C 58 68 02 01 CE 8A 40 01 31 19 00 00 F0 22 3B 61 05 00 09 15
```

Is there possibility that protocol in logger was changed?
`inverter_sn=175xxxx08` is my one.

MichaluxPL, 2022-05-11:
Where did You get those numbers from ? Attached SofarHYD.ES.ME3000SP.Modbus.protocol.pdf tells something totally different.
I guess configured range is much to big for HW data.
Please attach here MODBUS specs for Your inverter.
About changed logger protocol - I think somebody wrote something about it on elektroda.pl.

pixellos, 2022-05-11:
{
        "titleEN": "Software Version",
        "titlePL": "Wersja oprogramowania",
        "registers": ["0x0453","0x0454"],
        "optionRanges": [],
        "value_type": "SV"
      },
      This one is from 
      
[image]
```
 {
        "titleEN": "Hardware Version",
        "titlePL": "Wersja sprzętu",
        "registers": ["0x44D","0x44E"],
```
From 
[image]

MichaluxPL, 2022-05-11:
Yep, that's the one.
If You send me MODBUS specs, I can look at it and suggest something ... Your choice of course :)

pixellos, 2022-05-12:
I got it - it was mentioned in elektroda - different devices are having different codes - in this one is `0104`, for my one `0003`, now working on scraping an documentation XLSX to get all adresses and to clean up code a bit.

MichaluxPL, 2022-05-13:
This is exactly why I've suggested to get full MODBUS specs for Your device :)
Glad You've nailed it down ;)

MichaluxPL, 2022-05-16:
Did You manage to figure out the function number to read production data (for Your Inverter) ?

pixellos, 2022-05-16:
Just uploaded file with registries, if U want U can reuse it in Ur solution after changing format https://github.com/pixellos/codereinvented.automation.py/blob/master/cache/hyd3ph_g9_2021.json

closed 2022-05-25 by MichaluxPL