MichaluxPL/Sofar_LSW3 #49: azzurro HYBRID inverter can`t recieve value

r-schoepf, 2022-11-22:
we tried to run the script. Once with a Dietpi and once directly on the Synology, both without success.

We use an Azzurro hybrid inverter with the LSW-3 module.

Is there a possibility that we may ask for your help?

rimelis, 2022-11-22:
You can run your script in any device of your local network. 
It's another yet inverter with different address ranges and functional codes. If you have a coding skills in Python you have to find the correct description of modbus protocol suitable for your inverter (maybe this one: https://forum.iobroker.net/assets/uploads/files/1638097185191-modbus-communication-protocol-storage-and-hybrid-inverter.pdf) and try to play with various ranges watching what is coming from. I have found that the fork of this project https://github.com/pawsuch/Sofar_G3_LSW3 is more minimalistic and works better as a starting point.

r-schoepf, 2022-11-22:
tried it... error:

Traceback (most recent call last):
  File "/volume1/PYTHON/Sofar_LSW3/InverterHWData.py", line 47, in <module>
    crc=binascii.unhexlify(str(padhex(hex(libscrc.modbus(businessfield)))[4:6])+str(padhex(hex(libscrc.modbus(businessfield)))[2:4])) # CRC16modbus
AttributeError: module 'libscrc' has no attribute 'modbus'

r-schoepf, 2022-11-22:
an for the [Sofar_LSW3](https://github.com/MichaluxPL/Sofar_LSW3) we recieve the following error:

File "/root/Temp$/Sofar_LSW3-main/InverterData.py", line 164, in <module>
    data = clientSocket.recv(1024);

MichaluxPL, 2023-02-19:
> tried it... error:
> 
> Traceback (most recent call last): File "/volume1/PYTHON/Sofar_LSW3/InverterHWData.py", line 47, in crc=binascii.unhexlify(str(padhex(hex(libscrc.modbus(businessfield)))[4:6])+str(padhex(hex(libscrc.modbus(businessfield)))[2:4])) # CRC16modbus AttributeError: module 'libscrc' has no attribute 'modbus'

Looks like lack of some python libriaries installed or a different version, that lacks some subroutines.

rimelis, 2023-02-20:
If you have a plans to run this on Synology I suggest to create Docker container for this script and install everything you need inside it. Synology creates a lot of headache as it's really stripped OS.

closed 2023-09-21 by MichaluxPL