MichaluxPL/Sofar_LSW3 #52: Reading only 1 value

sibox1337, 2023-05-05:
Hello, I got some chinese inverter INVT BD6TKL-RL1
The problem is when i try to use the script it only gives me one single value and its -16143 and i think its kind of error that i cannot understand? Can u please help

MichaluxPL, 2023-05-05:
Response You get is much too short, so in fact You do not get any valuable data.
Unfortunately I don't know that inverter, so won't be able to help.
I'd suggest messing around with register addresses (best if You get it from the manufacturer), but there is no guarantee this script uses communication protocol suitable for Your inverter.

sibox1337, 2023-05-05:
Can i test with single register only for example 0x0404 so when i receive some that to know where to look for?

rimelis, 2023-05-05:
Each inverter has it's own table of registers. Even generations of the same model can differ a lot. For example Sofar Solar KTL-X G3 has absolutely different ranges than G2. You have very little chances to guess the right registers without documentation.

closed 2023-09-21 by MichaluxPL