MichaluxPL/Sofar_LSW3 #54: SS HYD 10

Kosa1978, 2023-06-13:
register_start1=0x0000          # Inverter register's first MODBUS address for the first register's range.
register_end1=0x0027            # Inverter register's last MODBUS address for the first register's range
register_start2=0x0105          # Inverter register's first MODBUS address for a second register's range
register_end2=0x0114            # Inverter register's last MODBUS address for a second register's range
registerhw_start=0x2000         # Like above, but for InverterHWData.py
registerhw_end=0x200D 
I'm having trouble entering records here. from sofar I have registers in a system other than hex how to replace it? which to choose for a given range 1,2,3. Do xml files need to be well-completed for the code to work without sending it to domoticz?

MichaluxPL, 2023-09-21:
If I understand You correctly, You have to convert register numbers (You have in Your docs) to hex values first. Than simply input it in config.cfg. XML file have to reflect Your register numbers, otherwise You won't get correct data (either You get values mapped to a wrong parameters or won't get any values at all).

ilia92, 2024-12-06:
I have complete solution for HYD - https://github.com/ilia92/sofar-monitor