MichaluxPL/Sofar_LSW3 #55: Inverter sometimes sending invalid data

miikaforma, 2023-09-22:
It seems like the inverter sometimes sends invalid data (rarely but happens) which results in this script logging wrong values. Unfortunately we can't stop the inverter from doing that. However we can choose to ignore the invalid responses (at least I think so). This has happened to me 3 times but unfortunately I only have the raw response from one of the cases. However I believe it gives enough knowledge for implementing a fix.

The issue comes from that the script expects the response to have valid 0x03 function payload. For some reason, I've gotten at least one response where there's actually two 0x04 function payloads in front of the requested 0x03 function payload 🤔 I've removed some "personal" data from the images but here's an example of the valid response and one invalid response.

Valid modbus payload:
[image]

Invalid modbus payload:
[image]

Possible things to fix the script from logging invalid values:
1. Start reading the data and if the modbus payload doesn't start with the 0x03 function code, stop the script (or re-request the data and try again _but remember to not create infinite loop in case the inverter keeps providing faulty data_).
2. Start reading the response and ignore the invalid 0x04 function payloads until the correct 0x03 function is reached (can easily skip since there's the length right after the function code, then just also skip the two bytes at the end (registerEnd and Crc16)). _The 0x03 function payload in my invalid response actually did have the correct updated data. Just it's not where the script expects it to be._
3. The comment [here](https://pysolarmanv5.readthedocs.io/en/latest/solarmanv5_protocol.html#header) on the Serial in the header is interesting but I'm not sure if it would work for this.

As a side note, while investigating this issue, I found this (https://github.com/kubaceg/sofar_g3_lsw3_logger_reader/issues/12) from another logger reader. I believe it's caused by this same invalid data.

kubaceg, 2023-10-27:
Thanks for spotting same problem in my logger. I'll make some additional log for bytes 26-27 to confirm corrupted function code. Hope this solution will have 100% of faulty readings cover because after my tests which was based on Faults there was some discrepencies where I had faulty readings without faults.

netadair, 2024-03-22:
The LSW-3 (or wifi sticks in general) seem to not mutex the RS485 bus in this multi master setup, so Modbus frames from the logger itself querying the inverter resp sending frames to portal are mixed with your requests/responses.

kalwinskidawid, 2024-06-04:
@netadair Did you resolve the problem with malformed packages?