MichaluxPL/Sofar_LSW3 #59: 0x0000 - Inverter status: -12863 0x0001 - Fault 1: -20715

Kedryn, 2024-05-14:
i have 10 LSW-3 
i tried your script against one, but i get

`Connecting to logger... connected successfully !
*** Chunk no:  0
Sent data:  bytearray(b'\xa5\x17\x00\x10E\x00\x00\xcc\r=\xa4\x02\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x03\x00\x00\x00(E\xd4\x00\x15')
Received data:  b'\xa5\x13\x00\x10\x15\x00r\xcc\r=\xa4\x02\x01\xb8\xb1\x8f\x00-&\x00\x00\xf7\xcd\xb3e\x01\x90\x02\xcd\xc1\xaf\x15'
0x0000 - Inverter status: -12863
0x0001 - Fault 1: -20715
No value in response for register 0x0002`

The LSW-3 firmware version is LSW3_15_270A_1.6

jirris, 2024-08-02:
This is likely for G3 inverter? There are some forks available for G3, but no XMLs. I have tried to modify config with the data from attached excel, but haven't got it working yet. Just getting error response from Inverter.

[SOFAR-G3.External.Modbus.Protocol-Engilsh-V211103.xlsx](https://github.com/user-attachments/files/16473865/SOFAR-G3.External.Modbus.Protocol-Engilsh-V211103.xlsx)

jirris, 2024-08-03:
I created a fork to support G3 as author preferred. Feel free to try: https://github.com/jirris/Sofar_LSW3_G3