MichaluxPL/Sofar_LSW3 #61: Error while running py

sheminasalam, 2024-08-12:
I get the following error.

```
Traceback (most recent call last):
  File "/home/pi/Sofar_LSW3/InverterData.py", line 315, in <module>
    client=paho.Client("inverter")
           ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/Sofar_LSW3/venv/lib/python3.11/site-packages/paho/mqtt/client.py", line 772, in __init__
    raise ValueError(
ValueError: Unsupported callback API version: version 2.0 added a callback_api_version, see docs/migrations.rst for details
```
hope you can help

rimelis, 2024-08-13:
It seems you have installed paho >= 2.0. It has one breaking change, decribed here:

https://stackoverflow.com/questions/77984857/paho-mqtt-unsupported-callback-api-version-error

sheminasalam, 2024-08-13:
I have downgraded to 1.6.1 and now the errors are gone. I also get the data output in json format in the terminal when I run the python file. but it seems even after that the program is not connecting to mqtt server or sending data. There are no error messages on the terminal also. why is this happening.

rimelis, 2024-08-13:
Have you changed mqtt_basic value from 0 to 1 in config.cfg file? If not then it will be no publish into MQTT. If you changed then you may add more WriteDebug calls to get better understanding what is happening.
This project is not the complete solution as you may already realize. It's not updated for a while. You may fork it and modify as you wish, it has been stated by author.

sheminasalam, 2024-08-13:
> Have you changed mqtt_basic value from 0 to 1 in config.cfg file? If not then it will be no publish into MQTT. If you changed then you may add more WriteDebug calls to get better understanding what [...]

this was it. so simple. thanks. 
yes

closed 2024-08-13 by sheminasalam